qmk/qmk_firmware #21351: Modifier with Mouse Button 1, resets after 3-5 times of pressing.

Justtheflu, 2023-06-23:
### Describe the Bug

I have a key: LGUI+Mouse Button 1, to select different layers in photoshop for instance.
Whenever I use it, after selecting the 4th or 5th separate layer, it'll reset and deselect everything. Like it forgot the put the modifier with the mouse button.

### Keyboard Used

keebio/bfo9000

### Link to product page (if applicable)

it's not made anymore

### Operating System

macos

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

Justtheflu, 2023-06-23:
managed a solution with bettertouchtool.

either way, weird behavior.

sigprof, 2023-06-23:
As a guess, maybe the problem is caused by the modifier report and the mouse report getting reordered, because they are sent over different USB endpoints (`keebio/bfo9000` does not enable NKRO by default, and the 6KRO keyboard reports are sent over a dedicated endpoint, while mouse reports go over the shared endpoint).  So you may try the following things:
- Rebuild the firmware with `NKRO_ENABLE=yes`, enable NKRO (with `NK_TOGG` or `NK_ON` keycodes), then repeat the test.  In this case the key events (including modifier state changes) would be sent in the NKRO report, which goes over the same shared endpoint, therefore the reordering of modifier and mouse reports won't happen.
- Rebuild the firmware with `KEYBOARD_SHARED_EP=yes`.  In this case the 6KRO keyboard reports will go over the same shared endpoint as the mouse reports, which should prevent any reordering.

Both solutions may result in some compatibility problems though (especially if you happen to use something like a KVM switch).

Justtheflu, 2023-06-24:
Thank you so much, this solved the issue.

closed 2023-06-24 by Justtheflu