qmk/qmk_firmware #21355: [Feature Request] `LM` + `TO` layer function

mcp292, 2023-06-23:
### Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [X] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

`LM` is useful but must be held. It would be nice to have a `TO` function that activates a layer with a mod.

This should reduce keymap code. For example if you have a numpad layer, instead of manually re-writing the numpad layer, replacing the numbers with shifted symbols, you can define a `TOM` key to modify your existing layer (e.g., `TOM(NUMPAD, MOD_LSFT)`).

myst729, 2023-06-25:
https://docs.qmk.fm/#/keycodes?id=us-ansi-shifted-symbols

sigprof, 2023-06-27:
Not sure whether the described use case is really useful enough to warrant reserving a bunch of keycodes.  Note that the 16-bit keycode space is far from infinite, and the proposed keycode range would require at least 4 bits for the layer number and 5 bits for the modifiers in order to be general enough, so a range of 512 consecutive keycode values would be needed.

You should be able to define a custom keycode that does `layer_move(_NUM); register_code(KC_LSFT);` on press; then you need some way to unregister the modifier — you may use the `layer_state_set_user()` callback to unregister the modifier when the layer gets turned off by any means, or maybe implement a toggling behavior for your custom keycode.

mcp292, 2023-06-28:
>  Note that the 16-bit keycode space is far from infinite...

> 4 bits for the layer number and 5 bits for the modifiers in order to be general enough, so a range of 512 consecutive keycode values would be needed.

`2**16 = 65536`. That should be able to accommodate 512 more keycodes no (`65536/512 = 128`)?

zvecr, 2023-06-29:
> 2**16 = 65536. That should be able to accommodate 512 more keycodes no (65536/512 = 128)?

That does not take into consideration whats already used. We have very few large concurrent blocks remaining.

> Not sure whether the described use case is really useful enough to warrant reserving a bunch of ke

I completely agree with this, and would rather reserve those keycode blocks for functionality that would be more widely used.

mcp292, 2023-06-29:
> That does not take into consideration whats already used. We have very few large concurrent blocks remaining.

That's what I was reaching for with the calculations. I was hoping someone would tell me, how much space is left?

> I completely agree with this, and would rather reserve those keycode blocks for functionality that would be more widely used.

If the interest is not there that's fair, but how do we find out?

getreuer, 2023-07-25:
Most of the larger used contiguous keycode ranges are spent in the "advanced" keycodes that take an argument or two. The arguments need to be packed into the keycode itself, costing a combinatorial number of codes. Here is an incomplete list, adding up to 54016 codes:

* 32768 codes (half of all the codes) for Unicode `UC(codepoint)` (15-bit arg)
* 8192 codes for modifier keycodes `LSFT(kc)`, `RALT(kc)`, etc. (5-bit arg + 8-bit arg)
* 8192 codes for mod-taps `MT(mod, kc)` (5-bit arg + 8-bit arg)
* 4096 codes for layer-taps `LT(layer, kc)` (4-bit arg + 8-bit arg)
* 512 codes for layer-mods `LM(layer, mod)` (4-bit arg + 5-bit arg)
* 256 codes for tap dance `TD(n)` (8-bit arg)

And this is far from considering everything.

I think your best bet is to invoke this behavior through function calls (`layer_move(_NUM); register_code(KC_LSFT);` like @sigprof suggested). Or if through new keycodes, find a way of achieving something similar in fewer codes. Say, if `TOM` were limited to registering a single mod key (instead of any same-hand chord of mods), that reduces the args to 3 bits (down from 5) plus 4 bits for the layer for a cost of 128 contiguous codes. Something like

```c
#define TOM(layer, mod) (QK_TO_MOD | (((layer)&0xF) << 3) | ((mod)&0x07))

// Example: TOM(NUM, KC_LSFT)
```

There's still separately the question of how interest folks are in this feature, but 128 codes would be more tolerable.

closed 2023-11-25 by github-actions[bot]