qmk/qmk_firmware #21378: [Feature Request] magnetic encoder sensors like MA730, AS5601, etc.

myst729, 2023-06-25:
### Feature Request Type

- [ ] Core functionality
- [X] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [X] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

I'd like to place an encoder on my device that doesn't have the physical **sense of scale** like Alps EC11. Previously I've tried two Honeywell SS41F sensors placed orthogonally. They work great.

[image]

But the accuracy depends on the **magnetic grid**. Its price is much more expensive than an ordinary Alps EC11 encoder if I want to get equivalent accuracy (usually 15 or 20 pulses per 360 degrees). Size is also a problem BTW.

[image]

So I tend to other magnetic sensor ICs. Recently I'm trying to get an [MA730GQ-Z](https://www.digikey.com/en/products/detail/monolithic-power-systems-inc/MA730GQ-Z/7690020) work. I connect following pins to an dev board running QMK.

[image]

When I constantly rotate the magnet clockwise, most of the time it works as expect but in certain area (in an about 30 degrees angle) it fires the CCW output. If rotate CCW, the opposite happens. Also, the chip has a very high accuracy of 12 bits (4096 per 360 degrees). When it's still, it may fire CW or CCW randomly due to interference.

Any help about the next step to try?

sigprof, 2023-06-25:
If you have the SPI interface connected, you may try adding some code to read the current angle over SPI (use `spi_write(0)` to read 8 bits twice, because the value that gets transmitted by `spi_read()` and `spi_receive()` is undefined in general — it is not guaranteed to be 0) and check what happens with that value.  The next step may be trying to read other registers, and then maybe even write them (note that a register write needs a subsequent delay, and actually saves the value in NVRAM with some limited amount of writes, so in your code you should read the register value first, and perform a write only when the desired value differs from the current value).  In particular, you may want to reduce the number of pulses per revolution for the ABZ interface to bring it more in line with QMK expectations.

As for actually using SPI as the encoder interface in QMK, there were some PRs to make the encoder interface more extensible (e.g., #17321, which unfortunately went stale), but currently there is no proper way to do that (`planck/rev7` uses a nasty hack which can't really be recommended).

Raallen89, 2023-06-26:
> ### Feature Request Type
>
> - [ ] Core functionality
> - [X] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
> - [X] Alteration (enhancement/optimization) of existing feature(s)
> - [ ] New [...]

closed 2023-11-18 by github-actions[bot]

myst729, 2023-11-25:
@sigprof I get what I want in Arduino through SPI on an mega32u4 dev board. The Arduino MA730 lib can read 8-bit or 16-bit value through builtin Arduino AVR SPI APIs, accurate and stable.

I also successfully set up SPI peripheral in QMK. However `spi_read()` returns very different values. It takes much longer time to be stable, and not guaranteed to be accurate. Although its code looks exactly the same as Arduino.

[image]

I'm afraid I can't go any further. I'm not a hardware guy 😢