qmk/qmk_firmware #21379: [Bug] Trying to add Oled Display to QMK keyboard. Getting issue on i2c_master.c file

nehalvxavier, 2023-06-25:
### Describe the Bug

While trying to add an OLED display to a QMK keyboard which uses an RP2040 microcontroller getting issues while trying to compile. 
[image]

### Keyboard Used

Custom Macro Keyboard 

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

sigprof, 2023-06-25:
You need to enable and configure I2C first: <https://docs.qmk.fm/#/i2c_driver?id=arm-configuration>

Specifically for RP2040, you need to enable either `RP_I2C_USE_I2C0` or `RP_I2C_USE_I2C1` in `mcuconf.h`, depending on which pins you are using for I2C; then you define `I2C_DRIVER` to either `I2CD0` or `I2CD1` in `config.h` and define `I2C1_SDA_PIN` and `I2C1_SCL_PIN` appropriately (the `1` in this name is a historical artifact and must be used always, even if you are using the I2C0 peripheral).

closed 2023-06-26 by nehalvxavier

bukil, 2024-06-24:
let me see,