qmk/qmk_firmware #21419: [Feature Request] Morse Code

IllustratedMan-code, 2023-06-30:
### Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [X] New behavior

### Description

## The vision
You can set a key/2 keys to act as a Morse code key. When you press this key, QMK will interpret the long/short key sequences into characters (if 2 keys are set, one is long, the other short). Things like delay for character separation could be configurable.

## Why this would be cool
- It would be great for learning Morse code!
- 1 key keyboards would actually be feasible (30 wpm is possible!)

## Implementation
I'm happy to try to implement this myself, but I would love it if someone could point me in the right direction on the source code. It should be simple to implement, but any tips would be great!

Stanley00, 2023-07-23:
@IllustratedMan-code There's one implementation here, which based on timing different between dit and dash. You can also add two custom keycodes for as QK_DIT and QK_DASH so the  `process_record_user` function would be a little bit simpler:

https://github.com/daliusd/qmk_firmware/blob/morsilka/keyboards/a_dux/keymaps/morsilka/keymap.c

myst729, 2023-07-25:
Interesting idea.

IllustratedMan-code, 2023-08-01:
@Stanley00 Thanks! Looks like I would put the implementation  in the `qmk/quantum` directory right?

Stanley00, 2023-08-07:
> @Stanley00 Thanks! Looks like I would put the implementation in the `qmk/quantum` directory right?

I would suggest you put the implementation in your keyboard keymap folder, the same as in the repos in shared above.

pochoi, 2023-08-09:
I am wondering whether you are thinking about ["Straight Key" or "Semi-Automatic Key"](https://morsecode.world/keys.html#:~:text=A%20%22bug%22%20or%20%22semi,left%20or%20to%20the%20right.). It would be interesting to see an implementation of "Semi-Automatic Key".

closed 2023-12-09 by github-actions[bot]