qmk/qmk_firmware #21428: [Bug] `qmk flash` doesn't detect bootloader mode

karb94, 2023-07-02:
### Describe the Bug

I used to be able to flash my custom firmware without any issues in my old Arch Linux system. Now I'm running NixOS and I just tried to flash my keyboard after compiling my firmware without any issues. When running `qmk flash` and pressing the physical RESET button on the PCB qmk doesn't recognise the bootloader: 
```
Flashing for bootloader: qmk-dfu
Bootloader not found. Make sure the board is in bootloader mode. See https://docs.qmk.fm/#/newbs_flashing
Trying again every 0.5s (Ctrl+C to cancel)
```

I know for a fact that the keyboard successfully enters the bootloader mode after pressing the physical RESET button because 1) I can't type anything after I press it and 2) I tried flashing the firmware in my MacBook Air using Toolbox and after pressing the RESET button it shows the following on the log panel:
```
Atmel DFU device connected: ATMEL Atm32U4DFU (03EB:2FF4:0000)
```

I have added all the udev rules from `util/udev/50-qmk.rules` to my system.

Any ideas on what could be causing this issue?

### Keyboard Used

keebio/iris/rev5

### Link to product page (if applicable)

https://keeb.io/collections/pre-built-keyboards/products/iris-keyboard-pre-built

### Operating System

NixOS

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.2
Ψ QMK home: /home/carles/projects/qmk_firmware
Ψ Detected Linux (NixOS 23.11 (Tapir)).
Ψ Git branch: master
Ψ Repo version: 0.15.23
Ψ - Latest master: 2022-02-20 11:40:31 +0000 (8f89854eb7) -- customized config.h and keymap.json
Ψ - Latest upstream/master: 2023-07-01 13:00:13 +0900 (000ee31e2c) -- [Keyboard] Add Varanidae (#21340)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2023-07-01 13:00:13 +0900 (000ee31e2c) -- [Keyboard] Add Varanidae (#21340)
Ψ - Common ancestor with upstream/develop: None
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 12.2.1
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.1
Ψ Found dfu-programmer version 0.7.2
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb16)
Ψ - lib/chibios-contrib: 2023-01-11 16:42:27 +0100 --  (a224be1)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b973)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f)
Ψ QMK is ready to go
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

karb94, 2023-07-02:
Thank you for the suggestion. Unfortunately that didn't work either:
```
Flashing for bootloader: atmel-dfu
Bootloader not found. Make sure the board is in bootloader mode. See https://docs.qmk.fm/#/newbs_flashing
 Trying again every 0.5s (Ctrl+C to cancel)....................
```

sigprof, 2023-07-02:
> I have added all the udev rules from `util/udev/50-qmk.rules` to my system.

How exactly did you do that?

Adding these rules to `services.udev.extraRules` does not actually work, because they get the priority of 99, but the rules which set `TAG+="uaccess"` need to have priority 50, so that they come before some other rules that actually detect that tag and perform the distro-dependent action to assign the permissions (through either `systemd-logind` or the `elogind` replacement for it).  However, the check in `qmk doctor` does not verify the priority and therefore does not detect that problem.

Instead of `services.udev.extraRules`, you need to use `services.udev.packages = [ pkgs.qmk-udev-rules ];` (you may get some older version of the rules this way, but for `atmel-dfu` or `qmk-dfu` that should be enough).  Alternatively, if you really need the most recent rules, you may do
```nix
  services.udev.packages = [
    (pkgs.writeTextFile {
      name = "qmk-udev-rules";
      destination = "/etc/udev/rules.d/50-qmk.rules";
      text = ''
        ... insert util/udev/50-qmk.rules here ...
      '';
    })
  ];
```

karb94, 2023-07-02:
Wow @sigprof I'm impressed. You knew exactly what I did wrong. I couldn't find much info on how I had to add the rules in NixOS so I used the `service.udev.extraRules` and since `qmk doctor` didn't complain I thought they were added correctly. Using the `services.udev.packages = [ pkgs.qmk-udev-rules ];` method solved my issue and it's good to know how to add the most recent rules for the future.

One last question. While I tried this method I got a bit too excited and forgot to use the `atmel-dfu` bootloader and instead the `qmk-dfu` bootloader was used. Everything seems to be working fine regardless. Should I bother changing to the `atmel-dfu` bootloader?

closed 2023-07-02 by karb94

sigprof, 2023-07-02:
For the firmware the differences between `atmel-dfu`, `lufa-dfu` and `qmk-dfu` are almost nonexistent (there are different defines for them, so you can distinguish between those variants in the code if you really want, but there is no code in the QMK core that handles those bootloader variants in a different way).  The only thing which is specific to `qmk-dfu` is that you can actually build that bootloader from QMK sources using the `bootloader` target (or build a combined firmware+bootloader file using the `production` target — such file may be useful for factory flashing when the boards are mass-produced).  So you don't need to change the bootloader to `atmel-dfu`.

The actual bootloader code is different, however:
- `atmel-dfu` is a proprietary bootloader by Atmel (the binaries are published, but no sources are available). New ATmega32U4 chips come from the factory with that bootloader already flashed (unless you get the ATmega32U4-RC variant — those chips come without any bootloader and need to be ISP flashed first, but once you flash the bootloader and set the fuses, they behave exactly like ATmega32U4 chips).
- `lufa-dfu` is the MIT-licensed code from the LUFA project which implements the same protocol as the `atmel-dfu` bootloader, and therefore the same `dfu-programmer` tool can be used for flashing the firmware.  It also implements an API for in-application flash programming which is different from `atmel-dfu`, but QMK does not use that API, so this difference does not really matter.  It also has the `SECURE_MODE` configuration option which prohibits most operations before executing the Erase Flash command (like `atmel-dfu` does); however, in `lufa-dfu` the `SECURE_MODE` option is disabled by default.
- `qmk-dfu` is a modification of `lufa-dfu` which adds some functions that are useful for QMK:
  - The USB manufacturer and product strings are set according to the keyboard configuration (with `Bootloader` added to the product string).
  - The bootloader can be configured to exit and start the firmware when some specific key is pressed (the key is specified by a pair of matrix pins), so that if you enter the bootloader accidentally, you can exit it without reconnecting the USB cable or running some flashing tool.
  - The bootloader can use a LED or a buzzer to indicate its activity if such hardware is present.
  - The secure mode is disabled, therefore QMK Toolbox is able to flash the EEPROM without erasing the firmware (this is not possible on `atmel-dfu`); this can be useful for some split boards which use EEPROM handedness.