qmk/qmk_firmware #21431: Adding support for the Keycron K1 Pro

WhyNotHugo, 2023-07-02:
I recently received a Keycron K1 Pro which advertises compatibility with QMK, but there's not firmware available for it (which honestly makes this compatibility pretty meaningless).

I'd like to contribute to this project and see if I can add support for this keyboard, but I honestly don't know where to start. I'm a software developer and can read/write C code but looking at the code for other keyboards (e.g.: `keyboards/keychron/q12/matrix.c`), I've no idea what I'm reading.

At this point, I'm trying to understand if compatibility with this device is something that I can potentially address, or if I'm better off returning the unit. Is this something that I can do, or will I need internal knowledge about the device that only the manufacturer would know? Do you have any guide or starting point? What does the process of adding a device look like?

Just to be clear: **I'm not asking "can you add support for this keyboard?", I'm asking "what would I need to do if I aspired to do this myself?".**

adophoxia, 2023-07-02:
There aren't any plans to add the K Pro/Q Pro boards to upstream QMK anytime soon. The problem is that the code KC uses to make BT work with QMK, specifically for the MCU they're using is too custom to where, unless they can tone it down to where it is in a state to where it can be mergable, you'll have to work off of KC's fork over in the `bluetooth_playground` branch for the time being.

closed 2023-07-02 by zvecr

WhyNotHugo, 2023-07-02:
I take it that it's this branch: https://github.com/Keychron/qmk_firmware/tree/bluetooth_playground

If I understood you correctly, the portion of the code that makes QMK work over bluetooth is too custom. Could this potentially be skipped for now and only merge in the bits necessary for flashing over USB?

In any case, I can work with that branch for now. I had failed to find it myself, so thanks a lot for the reference.

zvecr, 2023-07-02:
> Please refrain from asking maintainers to add support for specific keyboards -- it is unlikely they will have hardware available, and will not be able to help.
Your best bet is to take the initiative, add support, then submit a PR yourself.

WhyNotHugo, 2023-07-02:
@zvecr I read the README before opening this issue. I've even made it clear that I'm not asking for maintainers to add support for specific keyboards and put that in bold.

I don't understand what you're trying to tell me here. Did you simply reply to me without reading my question?

zvecr, 2023-07-02:
> Did you simply reply to me without reading my question?

I read your question. However we are no longer accepting issues pertaining to adding keyboard support.

You are welcome to join the [QMK Discord](https://discord.gg/Uq7gcHh).

WhyNotHugo, 2023-07-02:
Honestly, I was asking for guidance with the intent to contribute here. I'm sorry that this bothers you so much, I didn't expect this to be unwelcome.

Either way, I've managed to work around my issue.

zvecr, 2023-07-02:
You're welcome!