qmk/qmk_firmware #21459: [Bug] `split.main` from `info.json` is not working

numToStr, 2023-07-06:
### Issue Description

I was going through [keyboard.jsonschema](https://github.com/qmk/qmk_firmware/blob/master/data/schemas/keyboard.jsonschema) to configure the right-side as master. So, I changed [`split.main`](https://github.com/qmk/qmk_firmware/blob/master/data/schemas/keyboard.jsonschema#L611C1-L611C1) to `right` but it did not work. I looked through `generate/config_h.py` and it seems it's using `split.primary` field instead of `split.main`.
https://github.com/qmk/qmk_firmware/blob/63872f990b9ee1e0f9a6b480e771a2278cf90919/lib/python/qmk/cli/generate/config_h.py#L134

Not sure which place needs the fixing.

zvecr, 2023-07-06:
See #18254, for now just use the config.h options.

Also, you want to use the `Bug report` type next time when reporting issues.

zvecr, 2023-11-01:
Closing as `split.main` has been removed. For now, `config.h` options are still the way to configure this.

closed 2023-11-01 by zvecr