qmk/qmk_firmware #21532: [Bug]  is no longer honored for building all keyboards with the "-km" included

thepappas, 2023-07-13:
### Describe the Bug

Use of "-kb all" with `qmk compile` is not honored to build all keyboards that have the listed keymap.

Result of `qmk compile -j 0 -kb all -km thepappas`
```
usage: qmk.exe compile [-h] [-c] [-e ENV] [-j PARALLEL] [-n] [-km KEYMAP] [-kb KEYBOARD] [filename]
qmk.exe compile: error: argument -kb/--keyboard: invalid keyboard_folder value: 'all'
```
I have `thepappas` keymap directories in both `planck/keymaps/` and `crkbd/keymaps` in my fork/branch.

Though a run of `qmk compile -j 0 -kb all -km via` nets the same
```
usage: qmk.exe compile [-h] [-c] [-e ENV] [-j PARALLEL] [-n] [-km KEYMAP] [-kb KEYBOARD] [filename]
qmk.exe compile: error: argument -kb/--keyboard: invalid keyboard_folder value: 'all'
```
Most keyboards have a `keymaps/via` defined

### Keyboard Used

crkbd and planck

### Operating System

Windows 10

### qmk doctor Output

```text
$ qmk doctor -n
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.2
Ψ QMK home: D:/Users/jpappas/kbd/qmk_firmware
Ψ Detected Windows 10 (10.0.19045).
Ψ QMK MSYS version: 2.0.0
Ψ Git branch: thepappas
Ψ Repo version: 0.21.6
⚠ Git has unstashed/uncommitted changes.
Ψ - Latest thepappas: 2023-07-13 14:20:04 -0600 (3d4ae2c510) -- Thepappas: Add QMK MSYS to settings.json
Ψ - Latest upstream/master: 2023-07-13 00:14:23 +0100 (28d5cd1696) -- Bump JamesIves/github-pages-deploy-action from 4.4.2 to 4.4.3 (#21518)
Ψ - Latest upstream/develop: 2023-07-12 23:15:00 +0000 (6113857745) -- Merge remote-tracking branch 'origin/master' into develop
Ψ - Common ancestor with upstream/master: 2023-07-13 00:14:23 +0100 (28d5cd1696) -- Bump JamesIves/github-pages-deploy-action from 4.4.2 to 4.4.3 (#21518)
Ψ - Common ancestor with upstream/develop: 2023-07-13 00:14:23 +0100 (28d5cd1696) -- Bump JamesIves/github-pages-deploy-action from 4.4.2 to 4.4.3 (#21518)
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 12.2.0
Ψ Found avr-gcc version 12.2.0
⚠ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.
Ψ Found avrdude version 7.0
Ψ Found dfu-programmer version 1.0.0
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-01-11 16:42:27 +0100 --  (a224be15)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

tzarc, 2023-07-13:
You can use `qmk mass-compile -km thepappas` instead.

closed 2023-07-13 by tzarc

thepappas, 2023-07-13:
Ahh, then I need to submit a PR for docs/cli_commands.md reflecting this change. Correct?

tzarc, 2023-07-13:
Ahh, I didn't realise that was actually documented. I'll sort it out, there's probably some cleanup worth doing. Cheers though!

tzarc, 2023-09-22:
@thepappas apologies for it taking so long, but this should now be working. I know I gave the workaround previously, but the docs should now be consistent.