qmk/qmk_firmware #21556: [Bug] STM32 does not communicate over Serial

mhbkamaei, 2023-07-18:
### Describe the Bug

#Describe the Bug

I'm trying to build mitosis like keyboard with stm32f401 black pill and nrf52832 module, if I use pro micro and avr UART driver every thing is ok but when switch to black pill and use Serial driver no communication, I simulated the nrf behavior with a usb to tty adapter and a python script and it worked, it seems serial(USART) driver is not compatible with nrf and there is no uart driver for ARM like one is for AVR.

### Keyboard Used

mitosis

### Operating System

Linux

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

fauxpark, 2023-07-18:
> there is no uart driver for ARM like one is for AVR.

This is false. https://docs.qmk.fm/#/uart_driver
I have explained this to you on Discord already. There is some configuration you are not doing correctly.

BillTercel, 2023-07-27:
[Bug] STM32 does not communicate over Serial #21556
That's true.
I didn't test it on stm32f303，Both 103 and 401 failed the test, but the #include "hal_uart.h" serial port did.