qmk/qmk_firmware #21568: [Feature Request] Caps Word: Auto-press shift momentarily for each letter, instead of all the time

mellanslag-de, 2023-07-20:
### Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [X] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

Currently, when I enable Caps Word and type at least one letter, shift is held down for 5 seconds (until timeout).

When I e.g. click around in my IDE in that time frame, text fragments get selected (between old cursor position and new cursor position), because shift is held down.

I want to propose to not keep shift pressed the whole time, but instead momentarily enable it for each key press while caps word is on, and immediately after sending the pressed key, releasing shift again.

That way, if I click around, no unwanted behaviour is caused.

---

Was that already considered? Any implications I currently dont see? Or can we make it configurable?

mellanslag-de, 2023-07-20:
The original Issue where the feature was implemented: https://github.com/qmk/qmk_firmware/pull/16588

getreuer, 2023-07-25:
That's an interesting idea!

It's not what you asked for, but it might help that Caps Word's 5-second timeout can be configured ([documentation](https://docs.qmk.fm/#/feature_caps_word?id=idle-timeout)). Tighten it to, say, 3 seconds by adding in config.h:

```c
#define CAPS_WORD_IDLE_TIMEOUT 3000  // 3 seconds.
```

closed 2024-01-28 by github-actions[bot]