qmk/qmk_firmware #21590: [Bug] wuque/mammoth75x default layout cannot change keymaps for knob

DustRespirator, 2023-07-23:
### Describe the Bug

I have a mammoth75x M1 (wired PCB) and I use QMK configurator to generate firmware.

I chose LAYOUT_all, which is the only layout has knob turning key (matrix(row, col): (5,3) and (5,5))

The default keymap of this layout is KC_VOLD and KC_VOLU for layer 0, KC_MPRV and KC_MNXT for layer 1.

But no matter what keymap I set for knob turning, it will always be KC_VOLD and KC_VOLU for any layer enabled.

### Keyboard Used

wuque/mammoth75x

### Operating System

Windows 11

### qmk doctor Output

```text
I use QMK configurator so I do not have such output log
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Additional Context

Can I set VIA_ENABLE = yes when I use QMK configurator?

waffle87, 2023-07-25:
That code is guarded by an [`#ifdef VIA_ENABLE`](https://github.com/qmk/qmk_firmware/blob/master/keyboards/wuque/mammoth75x/mammoth75x.c#L19), which is not enabled when using the QMK Configurator, nor is it possible to enable. Currently, your best bet is to use VIA-enabled firmware or [compile the firmware yourself](https://docs.qmk.fm/#/newbs_getting_started) with the desired behaviour. It's worth mentioning that that code has been removed on the `develop` branch for reasons exactly like this. (#20389)

DustRespirator, 2023-07-25:
> Currently, your best bet is to use VIA-enabled firmware or [compile the firmware yourself](https://docs.qmk.fm/#/newbs_getting_started) with the desired behaviour. (#20389)

For some reasons (VIA does not support GUI_TOG) I may need to compile firmware. But compiling through QMK Configurator does not make sense for this problem even the default behavier was set for knob. So I may need to compile by myself.

I checked https://github.com/qmk/qmk_firmware/blob/master/keyboards/wuque/mammoth75x/rules.mk and https://docs.qmk.fm/#/feature_encoders?id=encoder-map I think what I should do is to add `ENCODER_MAP_ENABLE = yes` in rules.mk and set encoder in keymap.c? Setting encoder as a key is unavailable anymore, do I understand it correctly?

DustRespirator, 2023-07-27:
Thank you for your explanation! I'll do the rest on my own.