qmk/qmk_firmware #21633: [Bug] KC_CAPS swapped with LCTL

andmarti1424, 2023-07-28:
### Describe the Bug

Hello. I just bought a planck from DROP. v7.
I flashed my keymap but some of my keymappings are not working properly:
`MT(MOD_LCTL, KC_ESC)` for instance.

`C(KC_A)` is another one.
Seems LEFT CONTROL is swapped, cause if I change the `MT(MOD_LCTL, KC_ESC)`  to `MT(KC_CAPS, KC_ESC)` I get the expected behaviour.
`C(KC_A)` is not working at all.
Any ideas? Ill try to attach my keymap. Many thanks for your help!

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

andmarti1424, 2023-07-28:
[keymap.c.txt](https://github.com/qmk/qmk_firmware/files/12198733/keymap.c.txt)

Nebuleon, 2023-07-29:
You might have Caps Lock swapped with Control. Can you edit a key to `CL_TOGG` (Toggle Caps Lock and Left Control swap, documented here: https://docs.qmk.fm/#/keycodes?id=magic-keycodes) on one of your layers, reflash your firmware, press that key and try `C(A)` again?

(Another thing: `MT` only accepts a `MOD_*` in its first parameter. Your `MT(KC_CAPS, KC_ESC)` seems to only happen to work because it interprets `KC_CAPS` as another modifier, possibly `MOD_RCTL`. But it is not the way to go.)

andmarti1424, 2023-07-29:
@Nebuleon thanks for your reply. unfortunately CL_TOGG does not appear to make a change. Please see my updated keymap.
`MT(MOD_LCTL, KC_ESC)` does not work before or after CL_TOGG. 
C(KC_A) or C(KC_SPACE) (my tmux shortcut) does not work either
[keymap.c.txt](https://github.com/qmk/qmk_firmware/files/12207437/keymap.c.txt)
thanks again for your help.

andmarti1424, 2023-07-29:
I can confirm though, that the problem is related to my OS caps lock swap, cause if I comment this line on my Xorg config        `Option "XkbOptions" "ctrl:swapcaps,terminate:ctrl_alt_bksp"` my keymaps seems to work ok.
any other idea on how to swap it again in a layer without having to alter my Xorg configuration?
The previous suggestion does not work.
thanks.

sigprof, 2023-07-29:
You may add the `EE_CLR` keycode and use it to clear the whole EEPROM configuration.

`CL_TOGG` does not actually affect the modifier mask in `MT()` or keycodes with extra modifiers; the only magic feature that could affect `MOD_LCTL` is `keymap_config.swap_lctl_lgui`, which is toggled by `CG_LSWP` (or `CG_SWAP`, which toggles the swap for both left and right Ctrl/GUI).

sigprof, 2023-07-29:
Ah, `ctrl:swapcaps` will certainly break any uses of `MOD_LCTL`, and there is no way to make QMK send `KC_CAPS` instead of `KC_LCTL` in that case, because the modifier keycodes actually have dedicated bits in the keyboard report.  So you could either use `MOD_RCTL` everywhere instead of `MOD_LCTL`, or just remove `ctrl:swapcaps` from your X11 configuration and configure the QMK keymap to have keys in the desired locations instead.

andmarti1424, 2023-07-29:
Thanks. Will try those suggestions later in the afternoon and report back. Thanks for your help!

andmarti1424, 2023-07-29:
I had to do an EE_CLR and remove ctrl:swapcaps. now everything works great!! this can be closed now. thanks!!

closed 2023-07-29 by andmarti1424