qmk/qmk_firmware #21638: [Bug]  make: illegal argument to -j -- must be positive integer!

dostoievsky, 2023-07-29:
### Describe the Bug

I am trying to compile a keyboard but it keep throwing a make error:

```
root@playground:~/qmk_firmware # qmk compile -kb clueboard/66/rev3 -km default
Ψ Compiling keymap with make --jobs=1 clueboard/66/rev3:default

make: illegal argument to -j -- must be positive integer!

```

### Keyboard Used

Any keyboard

### Operating System

freebsd

### qmk doctor Output

```text
root@playground:~/qmk_firmware # qmk doctor
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.0.0
Ψ QMK home: /root/qmk_firmware
⚠ Unsupported OS detected: freebsd-13.2-release-amd64-64bit-elf
Ψ Git branch: master
Ψ - Latest master: 2023-07-28 22:54:32 +0800 (03b3840942) -- Updated info.json of c2 pro (#21581)
Ψ - Latest upstream/master: 2023-07-28 22:54:32 +0800 (03b3840942) -- Updated info.json of c2 pro (#21581)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2023-07-28 22:54:32 +0800 (03b3840942) -- Updated info.json of c2 pro (#21581)
Ψ - Common ancestor with upstream/develop: None
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 11.3.0
Ψ Found avr-gcc version 11.2.0
⚠ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.
Ψ Found avrdude version 7.1
Ψ Found dfu-programmer version 0.7.2
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-01-11 16:42:27 +0100 --  (a224be15)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Additional Context

Doctor says it is not a supported system, but I dont think it is related.

How can I change this --jobs config to something else?

Thanks for the atention.

sigprof, 2023-07-29:
> ⚠ Unsupported OS detected: freebsd-13.2-release-amd64-64bit-elf

The QMK build system requires GNU Make, and `make` on FreeBSD is probably a BSD Make.  You probably need to install GNU Make, so that it would be available as `gmake` — QMK CLI would use `gmake` if it is available (however, if you have the `MAKE` variable set in the environment for some reason, that has even higher priority).

dostoievsky, 2023-07-29:
Based in @sigprof comment 

I checked if i have a variable:
```
root@playground:~ # echo $MAKE
MAKE: Undefined variable.

```

Then installed gmake with pkg and created a alias:

```
root@playground:~ # which make
make: 	 aliased to /usr/bin/gmake

```

And it worked!

```
root@playground:~/qmk_firmware # qmk compile -kb clueboard/66/rev3 -km default
Ψ Compiling keymap with gmake --jobs=1 clueboard/66/rev3:default

Making clueboard/66/rev3 with keymap default

Generating: .build/obj_clueboard_66_rev3_default/src/info_deps.d                                    [OK]
avr-gcc (GCC) 11.2.0
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Generating: .build/obj_clueboard_66_rev3/src/info_config.h                                          [OK]
Generating: .build/obj_clueboard_66_rev3/src/default_keyboard.c                                     [OK]
Generating: .build/obj_clueboard_66_rev3/src/default_keyboard.h                                     [OK]
Compiling: keyboards/clueboard/66/rev3/rev3.c                                                       [OK]
Compiling: .build/obj_clueboard_66_rev3/src/default_keyboard.c                                      [OK]
Compiling: quantum/keymap_introspection.c                                                           [OK]
Compiling: quantum/quantum.c                                                                        [OK]
Compiling: quantum/bitwise.c                                                                        [OK]
Compiling: quantum/led.c                                                                            [OK]
Compiling: quantum/action.c                                                                         [OK]
Compiling: quantum/action_layer.c                                                                   [OK]
Compiling: quantum/action_tapping.c                                                                 [OK]
Compiling: quantum/action_util.c                                                                    [OK]
Compiling: quantum/eeconfig.c                                                                       [OK]
Compiling: quantum/keyboard.c                                                                       [OK]
Compiling: quantum/keymap_common.c                                                                  [OK]
Compiling: quantum/keycode_config.c                                                                 [OK]
Compiling: quantum/sync_timer.c                                                                     [OK]
Compiling: quantum/logging/debug.c                                                                  [OK]
Compiling: quantum/logging/sendchar.c                                                               [OK]
Compiling: quantum/bootmagic/magic.c                                                                [OK]
Compiling: quantum/matrix_common.c                                                                  [OK]
Compiling: quantum/matrix.c                                                                         [OK]
Compiling: quantum/debounce/sym_defer_g.c                                                           [OK]
Compiling: quantum/main.c                                                                           [OK]
Compiling: quantum/mousekey.c                                                                       [OK]
Compiling: quantum/color.c                                                                          [OK]
Compiling: quantum/rgblight/rgblight.c                                                              [OK]
Compiling: quantum/process_keycode/process_rgb.c                                                    [OK]
Compiling: quantum/backlight/backlight.c                                                            [OK]
Compiling: quantum/process_keycode/process_backlight.c                                              [OK]
Compiling: platforms/avr/drivers/ws2812_bitbang.c                                                   [OK]
Compiling: quantum/led_tables.c                                                                     [OK]
Compiling: quantum/process_keycode/process_magic.c                                                  [OK]
Compiling: quantum/send_string/send_string.c                                                        [OK]
Compiling: quantum/process_keycode/process_grave_esc.c                                              [OK]
Compiling: quantum/process_keycode/process_space_cadet.c                                            [OK]
Assembling: platforms/avr/xprintf.S                                                                 [OK]
Compiling: platforms/avr/printf.c                                                                   [OK]
[... 14 lines cut ...]
Compiling: lib/lufa/LUFA/Drivers/USB/Class/Common/HIDParser.c                                       [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Device_AVR8.c                                        [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/EndpointStream_AVR8.c                                [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Endpoint_AVR8.c                                      [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Host_AVR8.c                                          [OK]
[... cut ...]
Linking: .build/clueboard_66_rev3_default.elf                                                       [OK]
Creating load file for flashing: .build/clueboard_66_rev3_default.hex                               [OK]
Copying clueboard_66_rev3_default.hex to qmk_firmware folder                                        [OK]
Checking file size of clueboard_66_rev3_default.hex                                                 [WARNINGS]

 * The firmware size is approaching the maximum - 27842/28672 (97%, 830 bytes free)
root@playground:~/qmk_firmware # 
```

Thank you, closing this issue.

closed 2023-07-29 by dostoievsky

sigprof, 2023-07-29:
The alias is not really needed — it changes the behavior of the `make` command in your shell, but does not affect the behavior of other programs started from that shell.  What actually fixes things here is that `qmk` looks for `gmake` in `$PATH` and uses it in preference to `make` if found:
```
Ψ Compiling keymap with gmake --jobs=1 clueboard/66/rev3:default
```
The `make` alias would help if you ever run `make` directly instead of going through the `qmk` frontend (but it might also break things if you use it in some other place which actually expects BSD make).