qmk/qmk_firmware #21678: [Bug] USB Initialization Race Condition

rpn32, 2023-08-02:
### Describe the Bug

I have been using a custom split keyboard with Elite-C v4 (ATMega32U4) microcontrollers that communicate over serial using a USB-C cable.

With the latest QMK Firmware, the keyboard functions after flashing, but stops working after unplugging it from the computer and then plugging it back in.

Wireshark showed "GET DESCRIPTOR Response" failing with "Broken pipe (-EPIPE) (-32)" after the host sent a "GET DESCRIPTOR Request DEVICE" message.

I found that adding at least a 16ms delay to the USB_Init function in the LUFA USB drivers resolved the issue.

It seems as if QMK isn't finished setting up the device descriptor in memory by the time it is used in the get_usb_descriptor function of tmk_core, but I can't confirm it because I don't have a serial to usb adapter.

### Keyboard Used

Custom Split Keyboard

### Operating System

Debian GNU/Linux 12 (bookworm)

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.2
Ψ QMK home: /home/user/qmk_firmware
Ψ Detected Linux (Debian GNU/Linux 12 (bookworm)).
Ψ Git branch: master
Ψ Repo version: 0.21.6
⚠ Git has unstashed/uncommitted changes.
Ψ - Latest master: 2023-07-30 00:22:39 -0400 (14e14e9ab8) -- Correct "less than" to "up to" in squeezing_avr?id=layers (#21639)
Ψ - Latest upstream/master: 2023-08-02 11:05:00 -0600 (34a2f14274) -- [Keyboard] Add Bento Box macropad (#20521)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2023-07-30 00:22:39 -0400 (14e14e9ab8) -- Correct "less than" to "up to" in squeezing_avr?id=layers (#21639)
Ψ - Common ancestor with upstream/develop: None
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 12.2.1
Ψ Found avr-gcc version 5.4.0
Ψ Found avrdude version 7.1
Ψ Found dfu-programmer version 0.6.1
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-01-11 16:42:27 +0100 --  (a224be15)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

drashna, 2023-09-30:
Does setting `USB_SUSPEND_WAKEUP_DELAY` to 200 or the like help?

rpn32, 2023-09-30:
@drashna, when I originally ran into this issue I tried USB_SUSPEND_WAKEUP_DELAY, SPLIT_USB_TIMEOUT, SPLIT_USB_TIMEOUT_POLL and SPLIT_WATCHDOG_TIMEOUT with various values but nothing helped.

Today I was able to recreate the same issue with version 0.22.10 of QMK. None of the above settings help, but I am still able to get it to work by adding a 16ms delay to USB_Init() in lib/lufa/LUFA/Drivers/USB/Core/AVR8/USBController_AVR8.c.

alexey-ivanov-official, 2024-02-28:
@rpn32 have the same problem.
can you, please, show how exactly do you implement a delay?
i`m not in clang and [googled](https://stackoverflow.com/questions/3930363/implement-time-delay-in-c/3930477#3930477) solutions looks pretty ugly for me)

paprikman, 2026-11-06:
I do not think that this is "completed" in any way. The race condition is still relevant, especially with modern chips like RP2040. If I add a custom (workaround) watchdog to perform a soft reset, then keyboard is working after waking up the PC. This should be why other people report that setting extreme `USB_SUSPEND_WAKEUP_DELAY` values (up to 5000 ms) helps.