qmk/qmk_firmware #21696: QMK Security Discussion

dm17, 2023-08-05:
### Feature Request Type

- [ ] Core functionality
- [X] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

One interesting feature I've not seen explored much - which I'd like to make a thread on here for anyone to add to - is security, mitigations, and hardening.

There's a lot of opportunity for securing input devices against well known attacks like key loggers, malicious firmware, and evil maiden like attacks. I see QMK as an optimal place for such mitigations. For example, a keyboard's QMK configuration could comunicate with the host OS to notify the user if any changes have occured in the keyboard's firmware. There are also a number of ways to mitigate against keyloggers placed physically between a keyboard's keys and the computer's USB port. I've noticed a few QMK users have added "keyboard passwords" to unlock their keyboards as well.

Please add your thoughts, ideas and recommendations here!

whitehorsesoft, 2023-08-21:
Personally I am against this although I do appreciate the consideration of security implications..

First, I don't see keyboard firmware hardening as need for 99% of users. Of course others may comment differently on this - but for myself and my experience with colleagues and teams I would suspect not only is this not a need, it is not even a want.

Secondly, in my opinion this will make the source much more difficult to debug and maintain. A good security implementation would be something that gets into *everything.* By design, security should not be relegated to one tidy code block. It will interface with and perform oversight of most everything. Thus it will be an added mental burden to consider for most - if not all - bug reports or feature enhancements.

QMK already balances tons of features across different boards. I think it does an okay job, where most users of these keyboards can use it as a real value-add for their hardware. But this proposal will push people away over the long term.

whitehorsesoft, 2023-08-21:
Forgot to mention, for issues that are truly security issues with the firmware there is an existing way to address that: https://github.com/qmk/qmk_firmware/security

(To tie back to this discussion, I take "truly security issues" to be things like backdoors etc in the code. Which is different from what is being proposed in this issue.)

dm17, 2023-08-21:
> Secondly, in my opinion this will make the source much more difficult to debug and maintain. A good security implementation would be something that gets into _everything._ By design, security should [...]

I wouldn't want to suggest any hardening strategies that would make it harder to debug or maintain; I'm definitely not suggesting obfuscation (security via obscurity). I don't really care about the 1% more than the 99% because I don't see the 1% as needing it more than the 99% in the case that someone from the 99% group wanted to enable it (like they enable any other security feature). Perhaps the 99% would be a much larger 99% if companies weren't preventing QMK usage due to some bad security connotation; and mere discussions like this could help bring important details to light.

closed 2023-12-21 by github-actions[bot]