qmk/qmk_firmware #21700: [Feature Request] Any interest in Nordic platform support for board like NRF52840 dongle?

hantianjz, 2023-08-06:
### Feature Request Type

- [X] Core functionality
- [X] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

While trying to build a modified version of [mitosis](https://www.reddit.com/r/MechanicalKeyboards/comments/66588f/wireless_split_qmk_mitosis/) keyboard I instead of building a new USB receiver, I just used the NRF52840 dongle as the receiver, and I hacked QMK into my [project](https://github.com/hantianjz/27x2_mitosis/tree/main/receiver). Also the idea of having 2 separate MCU, one to run the wireless connection and another for QMK just seemed wasteful to me. This seems to be design with [redox](https://github.com/mattdibi/redox-keyboard/tree/master/redox-w) also.

Anyways my implementation was really meant as a proof of concept to show that I can get QMK to run on NRF52840, and work as wireless receiver at the same time, which turns out works reasonably well with basic features. I would like to implement this properly with nrf52 supported in QMK.

So I was wondering what is this community's stance of adding support for something like the NRF52840 dongle, or Nordic platform in general. Especially given there is ZMK. I am personally try to stay away from Zephyr in all things, but would trying to implement nordic support in QMK be a waste of my time?

Anyways, I am just a firmware engineer trying to find a long term hobbit project to take on, happy to just hack away at things in my own fork, but if there is an interest I can clean up my hack and get a proper implementation going and hopefully maybe try to get it into mainline.

fauxpark, 2023-08-06:
nRF is a bit of a problem due to licensing issues with their SDK. There is some work being done to add RIOT to QMK which should bring along nRF support.

hantianjz, 2023-08-06:
Ah that make sense, not surprised by that. Thanks for the quick reply.

closed 2023-08-06 by hantianjz