qmk/qmk_firmware #21702: [Bug] oled_set_brightness does not update slave keyboard

ojob, 2023-08-06:
### Describe the Bug

Hi,
I have a Kyria rev2, so a split keyboard, with an OLED screen model SSD1306, powered by two Elite PI.

In order to adjust OLED screen brightness, I just implemented function `process_record_user()` in my `keymap.c`, using custom keycodes and `oled_set_brightness()` function:

```c

enum custom_keycodes {
    OLED_BRI_MAX = SAFE_RANGE, // OLED screen brightness max
    OLED_BRI_MID,              // OLED screen brightness middle
    OLED_BRI_DIM,              // OLED screen brightness dimmed
};

#define OLED_BRIGHTNESS_MAX 255
#define OLED_BRIGHTNESS_MIDDLE 50
#define OLED_BRIGHTNESS_DIMMED 1

(…)

bool process_record_user(uint16_t keycode, keyrecord_t *record) {
    /* custom treatment
    <https://docs.qmk.fm/#/custom_quantum_functions>
    */
    switch (keycode) {
        case OLED_BRI_MAX:
            if (record->event.pressed) {
                oled_set_brightness(OLED_BRIGHTNESS_MAX);
            }
            return false;
        (…)
``` 

The custom keys are mapped on left half.

When I flash the keyboard (both halves), and uses these keys, only the master side OLED screen brightness is updated; slave side brightness does not change. (I tested with USB cable plugged on both sides).

Any idea about what I miss here?
Thanks for your help.

### Keyboard Used

kyria/rev2

### Operating System

Xubuntu 23.04

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.2
Ψ QMK home: /home/joel/dev/claviers/kyria
Ψ Detected Linux (Ubuntu 23.04).
Ψ Git branch: master
Ψ - Latest master: 2023-08-06 09:08:24 +0200 (468db147df) -- 2023-08-06.0: add OLED screen brightness keys
Ψ - Latest upstream/master: 2023-08-05 10:16:10 +1000 (049a4baec0) -- Contributing docs: add note to make sure repo is up to date (#21694)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2023-07-21 02:06:46 +0200 (422d502903) -- [Keyboard] Add EC Theca (#21233)
Ψ - Common ancestor with upstream/develop: None
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 12.2.1
Ψ Found avr-gcc version 5.4.0
Ψ Found avrdude version 6.3-20171130
Ψ Found dfu-programmer version 0.6.1
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-01-11 16:42:27 +0100 --  (a224be15)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Additional Context

I am quite a beginner with QMK, as I got my Kyria only a week ago.

I have used a PlanckEZ, but obviously 1) it doesn't have an embedded screen 2) ZSA made such a good job at polishing such aspects that I have no experience with the internals.

drashna, 2023-09-30:
it looks like `oled_set_brightness` sends the command directly to the oled, and no syncing occurs. 

You can use `oled_get_brightness` to get the current brightness, though. 

currently, the only way to sync this would be to use the custom data sync option.

ojob, 2023-10-02:
Thanks, understood.

So now is time for me to do my housework, reading the [Data Sync Options](https://github.com/qmk/qmk_firmware/blob/master/docs/feature_split_keyboard.md#data-sync-options), which seems close enough:

> ## Data Sync Options
> (...)
>
>     #define SPLIT_TRANSPORT_MIRROR
>
> (...) The purpose of this feature is to support cosmetic use of key events (e.g. RGB reacting to keypresses).

And as pointed, performance may well be much better using [Custom data sync](https://github.com/qmk/qmk_firmware/blob/master/docs/feature_split_keyboard.md#custom-data-sync-between-sides-idcustom-data-sync), and event dedicated to my use case.

Thanks again, I'll keep posting; this  issue may be retagged as non-bug :wink: 

"Cosmetic use" seems quite in line with my needs.

But reading further,

drashna, 2023-10-04:
split transport mirror doesn't handle all keycodes, just a small subset.

closed 2024-04-13 by github-actions[bot]