qmk/qmk_firmware #21707: [Bug] Niu mini compilation failed: unknown MCU 'atmega32u4' specified

Juliaria08, 2023-08-07:
### Describe the Bug

I tried to compile the default niu mini, and got the following output:
```
Ψ Compiling keymap with make --jobs=1 kbdfans/niu_mini:default

QMK Firmware 0.21.6
Making kbdfans/niu_mini with keymap default

Generating: .build/obj_kbdfans_niu_mini_default/src/info_deps.d                                     [OK]
avr-gcc (GCC) 4.3.3
Copyright (C) 2008 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Generating: .build/obj_kbdfans_niu_mini/src/info_config.h                                           [OK]
Generating: .build/obj_kbdfans_niu_mini/src/default_keyboard.c                                      [OK]
Generating: .build/obj_kbdfans_niu_mini/src/default_keyboard.h                                      [OK]
Compiling: keyboards/kbdfans/niu_mini/niu_mini.c                                                   cc1: error: unrecognized command line option "-std=gnu11"
unknown MCU 'atmega32u4' specified
Known MCU names:
   avr2
   at90s2313
   at90s2323
   at90s2333
   at90s2343
   attiny22
   attiny26
   at90s4414
   at90s4433
   at90s4434
   at90s8515
   at90c8534
   at90s8535
   avr25
   attiny13
   attiny2313
   attiny24
   attiny44
   attiny84
   attiny25
   attiny45
   attiny85
   attiny261
   attiny461
   attiny861
   attiny43u
   attiny48
   attiny88
   at86rf401
   avr3
   at43usb320
   at43usb355
   at76c711
   avr31
   atmega103
   avr35
   at90usb82
   at90usb162
   avr4
   atmega8
   atmega48
   atmega48p
   atmega88
   atmega88p
   atmega8515
   atmega8535
   atmega8hva
   at90pwm1
   at90pwm2
   at90pwm2b
   at90pwm3
   at90pwm3b
   avr5
   atmega16
   atmega161
   atmega162
   atmega163
   atmega164p
   atmega165
   atmega165p
   atmega168
   atmega168p
   atmega169
   atmega169p
   atmega32
   atmega323
   atmega324p
   atmega325
   atmega325p
   atmega3250
   atmega3250p
   atmega328p
   atmega329
   atmega329p
   atmega3290
   atmega3290p
   atmega32hvb
   atmega406
   atmega64
   atmega640
   atmega644
   atmega644p
   atmega645
   atmega6450
   atmega649
   atmega6490
   atmega16hva
   at90can32
   at90can64
   at90pwm216
   at90pwm316
   at90usb646
   at90usb647
   at94k
   avr51
   atmega128
   atmega1280
   atmega1281
   atmega1284p
   at90can128
   at90usb1286
   at90usb1287
   avr1
   at90s1200
   attiny11
   attiny12
   attiny15
   attiny28
 [ERRORS]
 |
 |
 |
make[1]: *** [builddefs/common_rules.mk:361: .build/obj_kbdfans_niu_mini_default/keyboards/kbdfans/niu_mini/niu_mini.o] Error 1
Make finished with errors
make: *** [Makefile:392: kbdfans/niu_mini:default] Error 1
```

### Keyboard Used

niu_mini

### Link to product page (if applicable)

I didn't even flash it, (I have sucessfully compiled and flashed it in the past)

### Operating System

Linux

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.2
Ψ QMK home: /home/julian/Documents/.tools/qmk/qmk_firmware
Ψ Detected Linux (Artix Linux).
⚠ Missing or outdated udev rules for 'apm32-dfu' boards. Run 'sudo cp /home/julian/Documents/.tools/qmk/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'gd32v-dfu' boards. Run 'sudo cp /home/julian/Documents/.tools/qmk/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'usbtinyisp' boards. Run 'sudo cp /home/julian/Documents/.tools/qmk/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
Ψ Git branch: master
Ψ Repo version: 0.21.6
Ψ - Latest master: 2023-08-07 03:13:20 +1000 (d52bafade3) -- Fix `binepad/bn009` and `durgod/dgk6x/hades_ansi` keymaps (#21701)
Ψ - Latest upstream/master: 2023-08-07 03:13:20 +1000 (d52bafade3) -- Fix `binepad/bn009` and `durgod/dgk6x/hades_ansi` keymaps (#21701)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2023-08-07 03:13:20 +1000 (d52bafade3) -- Fix `binepad/bn009` and `durgod/dgk6x/hades_ansi` keymaps (#21701)
Ψ - Common ancestor with upstream/develop: None
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 9.2.1
Ψ Found avr-gcc version 4.3.3
Ψ Found avrdude version 7.2
Ψ Found dfu-programmer version 1.0.0
Ψ Found dfu-util version 0.10-dev
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-01-11 16:42:27 +0100 --  (a224be15)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Other keyboard-related software installed

Not applicable

### Additional Context

I have sucessfully in the past compiled and flashed this keyboard using qmk.

fauxpark, 2023-08-07:
Your AVR toolchain looks extremely old:
```
Ψ Found avr-gcc version 4.3.3
```
I'm not sure how you would have gotten that version... Please follow the newbs guide.
https://docs.qmk.fm/#/newbs

Juliaria08, 2023-08-07:
I have read the guide and compiled the default one, but now I am failing to compile my not modified from last time keymap, and I get the following error message:
```
QMK Firmware 0.21.6
Making kbdfans/niu_mini with keymap jmjl

avr-gcc (GCC) 8.3.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: quantum/keymap_introspection.c                                                          In file included from ./keyboards/kbdfans/niu_mini/keymaps/jmjl/keymap.c:1,
                 from quantum/keymap_introspection.c:5:
./keyboards/kbdfans/niu_mini/keymaps/jmjl/keymap.c:12:183: error: 'KC_LSPO' undeclared here (not in a function); did you mean 'SC_LSPO'?
  KC_T, KC_Y, KC_U, KC_I, KC_O, KC_P, KC_BSPC, KC_ESC, KC_A, KC_S, KC_D, KC_F, KC_G, KC_H, KC_J, KC_K, KC_L, KC_SCLN, KC_QUOT, KC_LSPO, KC_Z, KC_X, KC_C, KC_V, KC_B, KC_N, KC_M, KC_COMM, KC_DOT, KC_MINS, KC_RSPC, LCTL_T(KC_BSLS), KC_LCBR, KC_LGUI, KC_LALT, MO(1), KC_SPC, MO(2), KC_RALT, KC_EQL, KC_RCBR, KC_SLSH),
                                                                                                                               ^~~~~~~

./.build/obj_kbdfans_niu_mini/src/default_keyboard.h:41:4: note: in definition of macro 'LAYOUT_planck_mit'
   {k2A, k2B, k2C, k2D, k2E, k2F, k2G, k2H, k2I, k2J, k2K, k2L}, \
    ^~~
./keyboards/kbdfans/niu_mini/keymaps/jmjl/keymap.c:12:260: error: 'KC_RSPC' undeclared here (not in a function); did you mean 'SC_RSPC'?
  KC_G, KC_H, KC_J, KC_K, KC_L, KC_SCLN, KC_QUOT, KC_LSPO, KC_Z, KC_X, KC_C, KC_V, KC_B, KC_N, KC_M, KC_COMM, KC_DOT, KC_MINS, KC_RSPC, LCTL_T(KC_BSLS), KC_LCBR, KC_LGUI, KC_LALT, MO(1), KC_SPC, MO(2), KC_RALT, KC_EQL, KC_RCBR, KC_SLSH),
                                                                                                                               ^~~~~~~

./.build/obj_kbdfans_niu_mini/src/default_keyboard.h:41:59: note: in definition of macro 'LAYOUT_planck_mit'
   {k2A, k2B, k2C, k2D, k2E, k2F, k2G, k2H, k2I, k2J, k2K, k2L}, \
                                                           ^~~
./keyboards/kbdfans/niu_mini/keymaps/jmjl/keymap.c:16:119: error: 'RESET' undeclared here (not in a function); did you mean 'RTSEN'?
  [4] = LAYOUT_planck_mit(KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_LSCR, RESET, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_TRNS, EEP_RST, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_TRNS, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO)
                                                                                                                       ^~~~~
./.build/obj_kbdfans_niu_mini/src/default_keyboard.h:40:9: note: in definition of macro 'LAYOUT_planck_mit'
   {k1A, k1B, k1C, k1D, k1E, k1F, k1G, k1H, k1I, k1J, k1K, k1L}, \
         ^~~
./keyboards/kbdfans/niu_mini/keymaps/jmjl/keymap.c:16:205: error: 'EEP_RST' undeclared here (not in a function); did you mean 'HF_RST'?
 O, KC_NO, KC_NO, KC_NO, KC_NO, KC_LSCR, RESET, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_TRNS, EEP_RST, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_TRNS, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO, KC_NO)
                                                                                                                               ^~~~~~~

./.build/obj_kbdfans_niu_mini/src/default_keyboard.h:41:9: note: in definition of macro 'LAYOUT_planck_mit'
   {k2A, k2B, k2C, k2D, k2E, k2F, k2G, k2H, k2I, k2J, k2K, k2L}, \
         ^~~
 [ERRORS]
 |
 |
 |
make[1]: *** [builddefs/common_rules.mk:361: .build/obj_kbdfans_niu_mini_jmjl/quantum/keymap_introspection.o] Error 1
Make finished with errors
make: *** [Makefile:392: kbdfans/niu_mini:jmjl] Error 1
```

Doctor:
```
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.2
Ψ QMK home: /home/julian/Documents/.tools/qmk/qmk_firmware
Ψ Detected Linux (Artix Linux).
Ψ Git branch: master
Ψ Repo version: 0.21.6
Ψ - Latest master: 2023-08-07 03:13:20 +1000 (d52bafade3) -- Fix `binepad/bn009` and `durgod/dgk6x/hades_ansi` keymaps (#21701)
Ψ - Latest upstream/master: 2023-08-07 03:13:20 +1000 (d52bafade3) -- Fix `binepad/bn009` and `durgod/dgk6x/hades_ansi` keymaps (#21701)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2023-08-07 03:13:20 +1000 (d52bafade3) -- Fix `binepad/bn009` and `durgod/dgk6x/hades_ansi` keymaps (#21701)
Ψ - Common ancestor with upstream/develop: None
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 13.2.0
Ψ Found avr-gcc version 8.3.0
Ψ Found avrdude version 7.2
Ψ Found dfu-programmer version 1.0.0
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-01-11 16:42:27 +0100 --  (a224be15)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go
```

fauxpark, 2023-08-07:
Some keycodes have been renamed since you wrote that keymap. https://docs.qmk.fm/#/keycodes
You might hit more errors once you fix those up.

Juliaria08, 2023-08-07:
> Some keycodes have been renamed since you wrote that keymap. https://docs.qmk.fm/#/keycodes You might hit more errors once you fix those up.

Thanks, after updating, it compiled flawlessly :)

closed 2023-08-07 by Juliaria08