qmk/qmk_firmware #21720: [Bug] buffer overflow / empty elf

LemeRus, 2023-08-09:
### Describe the Bug

```bash
qmk compile
Ψ Compiling keymap with gmake --jobs=1 jorne/rev1:LemeRus

QMK Firmware 0.21.6
Making jorne/rev1 with keymap LemeRus

Generating: .build/obj_jorne_rev1_LemeRus/src/info_deps.d                                           [OK]
avr-gcc (GCC) 5.4.0
Copyright (C) 2015 Free Software Foundation, Inc.

[…]

Linking: .build/jorne_rev1_LemeRus.elf                                                              [ERRORS]
 | 
 | collect2: fatal error: ld  terminated with signa 6 [Аварийный останов], core dumped
 | compilation terminated.
 | *** buffer overflow detected ***: terminated
 | 
gmake[1]: *** [builddefs/common_rules.mk:269: .build/jorne_rev1_LemeRus.elf] Error 1
Make finished with errors
gmake: *** [Makefile:392: jorne/rev1:LemeRus] Error 1
```

### Keyboard Used

jorne/rev1

### Operating System

Linux Mint Cinnamon 21 x86_64

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.2
Ψ QMK home: /media/lemerus/870 EVO/repos/qmk_firmware
Ψ Detected Linux (Linux Mint 21).
Ψ Git branch: master
Ψ Repo version: 0.21.6
Ψ - Latest master: 2023-08-07 03:13:20 +1000 (d52bafade3) -- Fix `binepad/bn009` and `durgod/dgk6x/hades_ansi` keymaps (#21701)
Ψ - Latest upstream/master: 2023-08-07 03:13:20 +1000 (d52bafade3) -- Fix `binepad/bn009` and `durgod/dgk6x/hades_ansi` keymaps (#21701)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2023-08-07 03:13:20 +1000 (d52bafade3) -- Fix `binepad/bn009` and `durgod/dgk6x/hades_ansi` keymaps (#21701)
Ψ - Common ancestor with upstream/develop: None
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.3.1
Ψ Found avr-gcc version 5.4.0
Ψ Found avrdude version 6.3-20171130
Ψ Found dfu-programmer version 0.6.1
Ψ Found dfu-util version 0.9
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-01-11 16:42:27 +0100 --  (a224be15)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Additional Context

My neofetch:
``` bash
OS: Linux Mint 21 x86_64 
Kernel: 5.15.0-78-generic 
Uptime: 2 days, 19 hours, 30 mins 
Packages: 2824 (dpkg), 4 (snap) 
Shell: bash 5.1.16 
Resolution: 2560x1440, 1440x2560 
DE: Cinnamon 5.4.12 
WM: Mutter 
WM Theme: Mint-Y 
Theme: Mint-Y-Dark [GTK2/3] 
Icons: HighContrast [GTK2/3] 
Terminal: gnome-terminal 
CPU: Intel Xeon E5-2680 v3 (24) @ 3.300GHz
GPU: NVIDIA GeForce RTX 2060 SUPER 
Memory: 7918MiB / 31929MiB 

```
I tried to read makefile and command (common_rules.mk:267) to run it by hands, but also received an error.
Rule from makefile:
``` makefile
# Note the obj.txt depeendency is there to force linking if a source file is deleted
265  %.elf: $(OBJ) $(MASTER_OUTPUT)/cflags.txt $(MASTER_OUTPUT)/ldflags.txt $(MASTER_OUTPUT)/obj.txt | $(BEGIN)
266  	@$(SILENT) || printf "$(MSG_LINKING) $@" | $(AWK_CMD)
> 267  	$(eval CMD=MAKE=$(MAKE) $(CC) $(ALL_CFLAGS) $(filter-out %.txt,$^) --output $@ $(LDFLAGS))
268  	@$(BUILD_CMD)
```
Error:
```bash
avr-gcc: error: unrecognized command line option «-fcommo»
```

sigprof, 2023-08-10:
Known bug — the AVR toolchain in Debian (and derivatives like Ubuntu) is broken in many non-English locales:

- #9357
  - https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1006223

As a workaround, you can set `LANG=en_US.UTF-8` before compiling.

closed 2023-08-10 by sigprof

LemeRus, 2023-08-10:
I tried `export LANG=en_US.UTF-8` but nothing changed.
```
Linking: .build/jorne_rev1_LemeRus.elf                                                              [ERRORS]
 | 
 | collect2: фатальная ошибка: ld завершилась по сигналу 6 [Аварийный останов], core dumped
 | компиляция прервана.
 | *** buffer overflow detected ***: terminated
 | 
gmake[1]: *** [builddefs/common_rules.mk:268: .build/jorne_rev1_LemeRus.elf] Ошибка 1
Make finished with errors
gmake: *** [Makefile:392: jorne/rev1:LemeRus] Ошибка 1
lemerus@lemerus-desktop:/media/lemerus/870 EVO/repos/qmk_firmware$ echo $LANG
en_US.UTF-8
```

sigprof, 2023-08-10:
Apparently `LANG=en_US.UTF-8` did not take effect, as the localized output is still there.  Maybe you have other environment variables exported that have higher priority than `LANG` — then you may need `export LC_ALL=en_US.UTF-8`, or maybe `export LANGUAGE=en`.

LemeRus, 2023-08-10:
I tried `export LC_ALL=en_US.UTF-8` few moments later, also zero effect, now tried `export LANGUAGE=en`, it seems helpful