qmk/qmk_firmware #21739: Opposite to LT(layer, kc) ?? Turn on layer when tapped, kc when held..

andmarti1424, 2023-08-11:
hello. 

I cannot seem to find an answer for "Turn on layer when tapped, kc when held" in here `https://docs.qmk.fm/#/keycodes`
My personal use case would be to switch to a layer NUM when tapped, and LALT when held..
Thank you!

filterpaper, 2023-08-26:
You can code the solution for your use case with [intercepting mod tap](https://docs.qmk.fm/#/mod_tap?id=intercepting-mod-taps).

glebsexy, 2023-08-26:
Indeed, that's what I ended up doing:

```
// hold for cmd, tap for layer
bool process_record_user(uint16_t keycode, keyrecord_t *record) {
    switch (keycode) {
        case LT(BASE, KC_RCMD):
            if (record->tap.count && record->event.pressed) {
                layer_move(BASE); // Intercept tap function to move to base
            } else if (record->event.pressed) {
                register_mods(MOD_MASK_GUI); // Intercept hold function to hold CMD
            } else if (!record->event.pressed) {
                unregister_mods(MOD_MASK_GUI); // Unpress CMD
            }
            return false;
    }
    return true;
}
```

filterpaper, 2023-08-26:
The keycode `LGUI_T()` will activate GUI/CMD on hold as a normal function. So you can use that and just add codes to intercept tap. Do change the status of this issue if the matter is resolved.

glebsexy, 2023-08-26:
Oh that's even better, thanks! Will change this in my keymap. This issue isn't mine though, I just wanted to chime in as I had the same question :)

andmarti1424, 2023-08-26:
Will try this and report back. Thanks

glebsexy, 2023-08-26:
Here is the updated code using the mod-tap feature:

```

bool process_record_user(uint16_t keycode, keyrecord_t *record) {
    switch (keycode) {
        // hold for cmd, tap for layer
        case RCMD_T(KC_NO):
            if (record->tap.count && record->event.pressed) {
                layer_move(BASE); // Intercept tap function to move to base
            } else {
                return true; // process hold normally
            }
            return false;
    }
    return true;
}
```

filterpaper, 2023-08-26:
> Here is the updated code using the mod-tap feature:
>
> ```
>
> bool process_record_user(uint16_t keycode, keyrecord_t *record) {
>     switch (keycode) {
>         // hold for cmd, tap for layer
>         case [...]

Try
```c

bool process_record_user(uint16_t keycode, keyrecord_t *record) {
    switch (keycode) {
        // hold for cmd, tap for layer
        case RCMD_T(KC_NO):
            if (record->tap.count && record->event.pressed) {
                layer_move(BASE); // Intercept tap function to move to base
                return false;
            }
            break;
    }
    return true;
}

closed 2023-12-25 by github-actions[bot]