qmk/qmk_firmware #21771: [Feature Request] swap_hands_on not used

jerviscui, 2023-08-17:
### Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [X] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

I see the function `swap_hands_on` is not used.
The function `process_action` changed the field `swap_hands` but did not use the function `swap_hands_on`, Although the method is a wrapper around the `swap_hands` change.

Then I added a user callback method `__attribute__((weak)) void swap_hands_toggle_user(void)` and triggered it in a method such as `swap_hands_on`. 
Like the following:

```c
void swap_hands_on(void) {
    swap_hands = true;
    swap_hands_toggle_user();
}

void swap_hands_off(void) {
    swap_hands = false;
    swap_hands_toggle_user();
}

void swap_hands_toggle(void) {
    swap_hands = !swap_hands;
    swap_hands_toggle_user();
}
```

filterpaper, 2023-08-26:
The `swap_hands_on()` function is intended for user callback. That is why they are explained in the user documentation page.

jerviscui, 2023-08-27:
> The `swap_hands_on()` function is intended for user callback. That is why they are explained in the user documentation page.

But my test `swap_hands_on()` is not being called.

filterpaper, 2023-08-27:
> > The `swap_hands_on()` function is intended for user callback. That is why they are explained in the user documentation page.
> 
> But my test `swap_hands_on()` is not being called.

If I understand your expectations correctly, `swap_hands_on()` will not be called by QMK. User `keymap.c` can call it to enable swap hands.

jerviscui, 2023-08-29:
I mean that SH_TOGG pressed does not trigger `swap_hands_on()`.

jerviscui, 2023-08-29:
Just like `layer_state_set_kb()` is triggered on layer switching.

filterpaper, 2023-08-29:
> I mean that SH_TOGG pressed does not trigger `swap_hands_on()`.

No it does not:
- If you need to trigger swap hands on a keymap, `SH_TOGG` can be used.
- If you need to trigger swap hands in a function like `process_record_user()`, `SH_TOGG` cannot be used. Instead you will have to use `swap_hands_toggle()`.

jerviscui, 2023-09-05:
I wanted to trigger my logic after triggering swap hands via the `SH_TOGG` button. 
But `swap_hands_on()` is not triggered.

closed 2024-01-06 by github-actions[bot]