qmk/qmk_firmware #21778: [Bug] KC_PLUS generates "=" instead "+" (shifted "=")

LemeRus, 2023-08-18:
### Describe the Bug

Default keymap use KC_PPLS to print "+". It works until I need not symbol "±" (Lvl3mod + Shift + KC_EQL).
When I tried to compile firmware with KC_PLUS instead KC_PPLS, I got kb without "+" on _LOWER layer at all.
Why can be so?

### Keyboard Used

jorne/rev1

### Operating System

Linux Mint Cinnamon 21 x86_64

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

LemeRus, 2023-08-19:
Now I fixed it with
```c
    case (PLS_LCT):
      if(record->tap.count && record->event.pressed) {
        tap_code16(KC_PLUS);
      }
      return false;
```
in process_record_user, but it's local fix :(

filterpaper, 2023-08-26:
This is not a bug. See this [caveat section](https://docs.qmk.fm/#/mod_tap?id=caveats) for details.

closed 2023-09-30 by drashna