qmk/qmk_firmware #21794: [Bug] RETRO_SHIFT doesn't work

VasilyKrainov, 2023-08-21:
### Describe the Bug

`rules.mk`:
`AUTO_SHIFT_ENABLE = yes`
`config.h`:
 `#define RETRO_SHIFT 1000`

Holding for a several seconds and releasing `GUI_T(KC_F)` shows nothing, but `F` should be shown

### Operating System

MacOS 13.5

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

IsaacElenbaas, 2023-09-25:
Did you add `GUI_T(KC_F)` to Auto Shift with [`get_custom_auto_shifted_key`](https://github.com/qmk/qmk_firmware/blob/master/docs/feature_auto_shift.md#auto-shift-per-key)? The docs were just updated to make it more clear that that is also a necessary step.

VasilyKrainov, 2023-09-25:
It helps, thank you so much!

closed 2023-09-25 by VasilyKrainov