qmk/qmk_firmware #21797: [Other] Caps Word ignores respect system keyboard layout

nzure, 2023-08-21:
### Issue Description

I have configured layer 0 to be a regular qwerty keymap and switch to the workman keyboard layout in my OS settings. This means that my semicolon key is in the top row and the letter 'I' in the home row.

However, when using Caps Word, pressing `KC_I` disables caps word. At the same time, pressing `KC_SCLN` does not. I suspect that this is because caps word only uses layer 0 and does not react to the system keymap.

An obvious fix is to add `KC_SCLN` to `caps_word_press_user()`, but I was wondering if there is a better way to do this without altering the layout on layer 0.

getreuer, 2023-08-23:
## Set host to QWERTY

Yes, if it's an option, the easy solution is to do what @lesshonor says and set the OS to QWERTY, then arrange the QMK keymap with your desired layout.

## Use WK_ keycodes

If you must have Workman layout set on the OS, an alternative is to add at the top of your keymap.c:

```c
#include "keymap_workman.h"
#include "sendstring_workman.h"
```

The [keymap_workman.h](https://github.com/qmk/qmk_firmware/blob/master/quantum/keymap_extras/keymap_workman.h) header defines "`WK_`"-prefixed keycodes that account for the OS mapping scancodes to Workman.

Then, the critical step, define `caps_word_press_user()` described in [Configure which keys are “word breaking”](https://docs.qmk.fm/#/feature_caps_word?id=configure-which-keys-are-word-breaking) in terms of the `WK_` keycodes. Something like

```c
bool caps_word_press_user(uint16_t keycode) {
    switch (keycode) {
        // Keycodes that continue Caps Word, with shift applied.
        case WK_A:
        case WK_B:
        case WK_C:
        case WK_D:
        // and so on for all the letters...
        case WK_Z:
            add_weak_mods(MOD_BIT(KC_LSFT));  // Apply shift to next key.
            return true;

        // Keycodes that continue Caps Word, without shifting.
        case KC_1 ... KC_0:
        case KC_BSPC:
        case KC_DEL:
            return true;

        default:
            return false;  // Deactivate Caps Word.
    }
}
```

nzure, 2023-09-19:
@lesshonor I see, I created a new layer and I'm using that one now.

@getreuer Thank you very much for pointing that out to me. I replaced `KC_A ... KC_Z` with `WK_A ... WK_Z` and it works as expected now.

Thank you both very much for your help!

closed 2023-09-19 by nzure

Apteryks, 2023-09-28:
@nzure not that unless all the key codes between WK_A and WK_Z are contiguous, using a range case like `WK_A ... WK_Z` is wrong.