qmk/qmk_firmware #21810: [Bug] lily58:hvp asks for non-existent file

cupofjoakim, 2023-08-22:
### Describe the Bug

According to the instructions in lily58/hvp, compilation should be done by running `make lily58:hvp:dfu`. 

When I run this in QMK MSYS I get this output:
```sh
[cupofjoakim@pop-silent qmk_firmware]$ make lily58:hvp:dfu
QMK Firmware 0.21.6
Making lily58/rev1 with keymap hvp and target dfu

avr-gcc.exe (GCC) 8.5.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/lily58/lib/rgb_state_reader.c                                                  [OK]
Compiling: keyboards/lily58/lib/layer_state_reader.c                                                [OK]
Compiling: keyboards/lily58/lib/logo_reader.c                                                       [OK]
Compiling: keyboards/lily58/lib/keylogger.c                                                         [OK]
Compiling: users/hvp/hvp.c                                                                          [OK]
Compiling: users/hvp/tap_dances.c                                                                   [OK]
Compiling: .build/obj_lily58_rev1/src/default_keyboard.c                                            [OK]
Compiling: quantum/keymap_introspection.c                                                          In file included from quantum/keymap_introspection.c:5:
./keyboards/lily58/keymaps/hvp/keymap.c:9:12: fatal error: ssd1306.h: No such file or directory
   #include "ssd1306.h"
            ^~~~~~~~~~~
compilation terminated.
 [ERRORS]
 |
 |
 |
make[1]: *** [builddefs/common_rules.mk:361: .build/obj_lily58_rev1_hvp/quantum/keymap_introspection.o] Error 1
Make finished with errors
make: *** [Makefile:392: lily58:hvp:dfu] Error 1
``` 

If I understand what has happened properly, it seems there _was_ a file called `ssd1306.h` at one point, but that this has been replaced in most keymaps by a wrapper for oled drivers. I'm very new to both C and QMK so I'm not quite sure how to fix this code-wise.

TLDR: Layout is outdated and need updating.

### Keyboard Used

lily58

### Operating System

windows

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: C:/Users/joaki/qmk_firmware
Ψ Detected Windows 10 (10.0.22621).
Ψ QMK MSYS version: 1.7.2
Ψ Git branch: master
Ψ Repo version: 0.21.6
⚠ Git has unstashed/uncommitted changes.
Ψ - Latest master: 2023-08-21 22:47:46 +0100 (3f192722b1) -- Initial support for SHIFT v2 (#21756)
Ψ - Latest upstream/master: 2023-08-21 22:47:46 +0100 (3f192722b1) -- Initial support for SHIFT v2 (#21756)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2023-08-21 22:47:46 +0100 (3f192722b1) -- Initial support for SHIFT v2 (#21756)
Ψ - Common ancestor with upstream/develop: None
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.1.0
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.0
Ψ Found dfu-programmer version 0.7.2
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-01-11 16:42:27 +0100 --  (a224be15)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

HiryKun, 2023-08-28:
Obviously it's old code. It still use ole OLED APIs. But don't worry, the repair will be  merged in master branch recently.[More information](https://github.com/qmk/qmk_firmware/pull/21651)
If you don't want to wait for it, you can referto [the PR](https://github.com/qmk/qmk_firmware/pull/21651) and delete the old code.

HiryKun, 2023-08-29:
Obviously it's old code. It still use ole OLED APIs. But don't worry, the repair will be  merged in master branch recently.[More information](https://github.com/qmk/qmk_firmware/pull/21651)
If you don't want to wait for it, you can referto [the PR](https://github.com/qmk/qmk_firmware/pull/21651) and delete the old code.

Develop branch was merged in master. You can try to compile angin.
But I found that the original display task seems to be broken. If you find the display don't work, please report to maintainer.

closed 2023-11-01 by zvecr