qmk/qmk_firmware #21853: [Feature Request] unified approach to user secrets

3geek14, 2023-08-28:
### Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [X] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

Currently, `user_song_list.h` is in the root `.gitignore` file, so that individuals can more easily add custom songs to their boards without committing them to the main repo. As seen in PRs like #21821 and #21641, users have more than just songs to keep out of the repo. On quick inspection, there are 23 additional `.gitignore` files, about half of which reference some form of secrets.

I propose adding something to the root `.gitignore` file more generic in name than `user_song_list.h`. For consistency, I'd suggest `user_*.h` and `user_*.c`, though one existing file[1] would want to get renamed. Looking at existing ignore files, `secrets.h` and `secrets.c` are the most popular names to use, so perhaps that's a better thing to make official.

If people agree with this change, I volunteer to add documentation for it.

[1]: `keyboards/thevankeyboards/minivan/keymaps/josjoha/user_config.h`

elpekenin, 2023-08-28:
Seems kind of pointless given that QMK will stop accepting user keymaps and probably remove existing ones from the `qmk_firmware` repo. 

[Source](https://github.com/qmk/qmk_firmware/pull/21848/files#diff-91b6e96a12dfe82e00f3ef48e01758effb39587494d91ca1faa87408690aa7adR7-R9).

3geek14, 2023-08-29:
Ah, I hadn't realized just how soon that was landing. Closing this now.

closed 2023-08-29 by 3geek14