qmk/qmk_firmware #21867: Please add support for a UK '£' symbol

grab-a-byte, 2023-08-29:
### Feature Request Type

- [X] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

I've searched for Keycodes or anything for a brittish poound symbol and I am unable to find anything. As a uk based individual this feels like a bit of an oversight. I mainly configure using the VIA app as I am a newbie however I know it comes from capabilities in this firmware. Is it possible to get this added or is there a way to be able to configure this? 
Many Thanks in advance.

dunk2k, 2023-08-30:
As a fellow 'UK English' user I can confirm that the keycodes `KC_3`, when shifted, and `KC_HASH` result in a `£` character inputted when operating system's keyboard language is set to `UK English`.

US English key names are used as a default, an anchor point if you will, for abstracting the underlying (non-human language specific) code names for each key. 

Point to note also, Right Alt keycode(s) translate to Alt Gr key; this applies to all (operating system) keyboard languages that utilise the Alt Gr key.

grab-a-byte, 2023-09-01:
Ok, thanks for the heads up folks.

closed 2023-09-01 by grab-a-byte