qmk/qmk_firmware #21958: [Bug] keyboard info.json vs. user keymap.json config precedence

rhruiz, 2023-09-08:
### Describe the Bug

Should config in a user keymap.json be able to override features in the keyboard's info.json?

My scenario: I was trying to disable `rgblight` by setting `config.features.rgblight` to `false` and change the bootlader from `caterina` to `qmk-dfu` by setting `config.bootloader` to `qmk-dfu` to match how my promicros are flashed.

Since the qmk cli generates both the keyboard's and my keymap's `rules.mk` with directives like `BOOTLOADER ?= qmk-dfu` and the keyboard-level `rules.mk` are loaded before my keymap's, my keymap-level rules do not override the keyboard value.

This is the opposite behaviour of when I use my own keymap.c+rules.mk with directives like `BOOTLOADER = qmk-dfu` and `RGBLIGHT_ENABLE = no`

### Keyboard Used

crkbd/rev1

### Operating System

macOS 12.6

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.2
Ψ QMK home: /Users/ricardo/dev/keyboard/qmk_firmware
Ψ Detected macOS 12.6.8 (Intel).
Ψ Git branch: master
Ψ Repo version: 0.22.2
Ψ - Latest master: 2023-09-07 13:53:16 -0700 (3e3320e8d9) -- Cipulot EC Theca Community Layout Support (#21913)
Ψ - Latest upstream/master: 2023-09-07 13:53:16 -0700 (3e3320e8d9) -- Cipulot EC Theca Community Layout Support (#21913)
Ψ - Latest upstream/develop: 2023-09-07 20:53:59 +0000 (ddb737e716) -- Merge remote-tracking branch 'origin/master' into develop
Ψ - Common ancestor with upstream/master: 2023-09-07 13:53:16 -0700 (3e3320e8d9) -- Cipulot EC Theca Community Layout Support (#21913)
Ψ - Common ancestor with upstream/develop: 2023-09-07 13:53:16 -0700 (3e3320e8d9) -- Cipulot EC Theca Community Layout Support (#21913)
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.3.1
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.2
Ψ Found dfu-programmer version 1.1.0
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-07-17 11:39:05 +0200 --  (da78eb37)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Additional Context

the keymap json I tried: https://github.com/rhruiz/qmk_keymaps/blob/main/crkbd.json

zvecr, 2023-09-09:
#21960 should hopefully fix the issue.

rhruiz, 2023-09-09:
Worked here :)

PR #21960 (2023-09-13): Generate keymap.json config options more forcefully

## Description

Changes keymap level generation from:

```c
#ifndef BOOTMAGIC_LITE_COLUMN
#    define BOOTMAGIC_LITE_COLUMN 1
#endif // BOOTMAGIC_LITE_COLUMN
```

```make
BOOTLOADER ?= qmk-dfu
```

to
```c
#undef BOOTMAGIC_LITE_COLUMN
#define BOOTMAGIC_LITE_COLUMN 1
```

```make
BOOTLOADER = qmk-dfu
```

This should ensure that configuration gets correctly applied.

## Types of Changes

- [ ] Core
- [x] Bugfix
- [ ] New feature
- [ ] Enhancement/optimization
- [ ] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or updat [... cut]
closed 2023-09-13 by zvecr