qmk/qmk_firmware #21971: [Bug] CONVERT_TO=kb2040 causes error: implicit declaration of function 'LAYOUT'

Dashyyy, 2023-09-10:
### Describe the Bug

When trying to compile specifically with the option CONVERT_TO=kb2040 the error:
error: implicit declaration of function 'LAYOUT'
occurs, it's also followed by 
note: (near initialization for 'keymaps[1][0][0]')
error: initializer element is not constant

### Keyboard Used

redox_w (removed 1 row, added encoder)

### Link to product page (if applicable)

-

### Operating System

Linux, Windows

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.2
Ψ QMK home: /home/julian/qmk_firmware
Ψ Detected Linux (Ubuntu 22.04.3 LTS).
Ψ Git branch: master
Ψ Repo version: 0.22.2
Ψ - Latest master: 2023-09-02 23:34:09 -0500 (8ad43a965f) -- Added kiserdesigns keyboard folder and Madeline keyboard (#21857)
Ψ - Latest upstream/master: 2023-09-09 12:15:57 -0700 (9a29ddf6af) -- dnworks FRLTKL Community Layout support (#21922)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2023-09-02 23:34:09 -0500 (8ad43a965f) -- Added kiserdesigns keyboard folder and Madeline keyboard (#21857)
Ψ - Common ancestor with upstream/develop: None
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.3.1
Ψ Found avr-gcc version 5.4.0
Ψ Found avrdude version 6.3-20171130
Ψ Found dfu-programmer version 0.6.1
Ψ Found dfu-util version 0.9
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-07-17 11:39:05 +0200 --  (da78eb37)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Other keyboard-related software installed

-

sigprof, 2023-09-10:
If `redox_w` is not a typo, that code currently won't compile when using any of the RP2040 converters, because the UART driver (which that board uses to communicate with the wireless controllers) currently does not support RP2040 (although the error would be different from what you show).

However, your modifications like “removed 1 row, added encoder” may easily cause similar errors; without seeing your code I can't really advise anything about that.

Dashyyy, 2023-09-10:
yeah, so it turns out, it was a mistake in info.json

I saw #17744 that there is issue regarding the uart driver, are there any updates for that?

azhizhinov, 2024-05-02:
Hello.
We trying to replace ProMicro with RP2040-Zero.
The hardware part is ready, but I can't compile the firmware.
Can someone help with the firmware?
[20240227.REDOX_W.00.zip](https://github.com/qmk/qmk_firmware/files/15186460/20240227.REDOX_W.00.zip)

closed 2024-05-02 by fauxpark