qmk/qmk_firmware #22012: [Bug] various build failures building the ergodox_ez keyboard firmware

Apteryks, 2023-09-14:
### Describe the Bug

I'm attempting to build qmk_firmware using something like:
```
make ergodox_ez:all:all
```

And see the following failures:
```
Making ergodox_ez/base with keymap belak and target all                                        [ERRORS]
In file included from quantum/keymap_introspection.c:5:
./layouts/community/ergodox/belak/keymap.c: In function ‘process_record_user’:
./layouts/community/ergodox/belak/keymap.c:270:13: error: implicit declaration of function ‘process_unicode’; did you mean ‘process_record’? [-Werror=implicit-function-declaration]
             process_unicode((0x00AF|QK_UNICODE), record);   // Hand
             ^~~~~~~~~~~~~~~
             process_record
cc1: all warnings being treated as errors
 | 
 | 
 | 
make[1]: *** [builddefs/common_rules.mk:361: .build/obj_ergodox_ez_base_belak/quantum/keymap_introspection.o] Error 1
make[1]: *** Waiting for unfinished jobs....
[...]

users/danielo515/danielo515.c: In function ‘leader_start_user’:
users/danielo515/danielo515.c:7:6: error: type of ‘vodi’ defaults to ‘int’ [-Werror=implicit-int]
 void leader_start_user(vodi) {
      ^~~~~~~~~~~~~~~~~
users/danielo515/danielo515.c:7:1: error: number of arguments doesn’t match prototype
 void leader_start_user(vodi) {
 ^~~~
In file included from quantum/quantum.h:97,
                 from users/danielo515/danielo515.h:2,
                 from users/danielo515/danielo515.c:1:
quantum/leader.h:17:6: error: prototype declaration
 void leader_start_user(void);
      ^~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
 | 
 | 
 | 
make[1]: *** [builddefs/common_rules.mk:361: .build/obj_ergodox_ez_base_danielo515/danielo515.o] Error 1
make[1]: *** Waiting for unfinished jobs....

[...]

Making ergodox_ez/base with keymap default_glow and target all                                 [ERRORS]
quantum/keymap_introspection.c: In function ‘keymap_layer_count_raw’:
quantum/keymap_introspection.c:17:49: error: ‘keymaps’ undeclared (first use in this function)
 #define NUM_KEYMAP_LAYERS_RAW ((uint8_t)(sizeof(keymaps) / ((MATRIX_ROWS) * (MATRIX_COLS) * sizeof(uint16_t))))
                                                 ^~~~~~~
quantum/keymap_introspection.c:20:12: note: in expansion of macro ‘NUM_KEYMAP_LAYERS_RAW’
     return NUM_KEYMAP_LAYERS_RAW;
            ^~~~~~~~~~~~~~~~~~~~~
quantum/keymap_introspection.c:17:49: note: each undeclared identifier is reported only once for each function it appears in
 #define NUM_KEYMAP_LAYERS_RAW ((uint8_t)(sizeof(keymaps) / ((MATRIX_ROWS) * (MATRIX_COLS) * sizeof(uint16_t))))
                                                 ^~~~~~~
quantum/keymap_introspection.c:20:12: note: in expansion of macro ‘NUM_KEYMAP_LAYERS_RAW’
     return NUM_KEYMAP_LAYERS_RAW;
            ^~~~~~~~~~~~~~~~~~~~~
quantum/keymap_introspection.c: At top level:
quantum/keymap_introspection.c:17:49: error: ‘keymaps’ undeclared here (not in a function)
 #define NUM_KEYMAP_LAYERS_RAW ((uint8_t)(sizeof(keymaps) / ((MATRIX_ROWS) * (MATRIX_COLS) * sizeof(uint16_t))))
                                                 ^~~~~~~
quantum/keymap_introspection.c:30:16: note: in expansion of macro ‘NUM_KEYMAP_LAYERS_RAW’
 _Static_assert(NUM_KEYMAP_LAYERS_RAW <= MAX_LAYER, "Number of keymap layers exceeds maximum set by LAYER_STATE_(8|16|32)BIT");
                ^~~~~~~~~~~~~~~~~~~~~
quantum/keymap_introspection.c:30:41: error: ‘MAX_LAYER’ undeclared here (not in a function)
 _Static_assert(NUM_KEYMAP_LAYERS_RAW <= MAX_LAYER, "Number of keymap layers exceeds maximum set by LAYER_STATE_(8|16|32)BIT");
                                         ^~~~~~~~~
[... 967 lines cut ...]
Ψ Found avrdude version 6.3
Ψ Found dfu-programmer version 0.7.2
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-07-17 11:39:05 +0200 --  (da78eb37)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ Major problems detected, please fix these problems before proceeding.
Ψ Check out the FAQ (https://docs.qmk.fm/#/faq_build) or join the QMK Discord (https://discord.gg/Uq7gcHh) for help.
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Additional Context

I'm trying to build the firmware files for all the ergodox_ez keyboard to make them available as a package for GNU Guix.

zvecr, 2023-09-14:
Given the recent policy change, <https://docs.qmk.fm/#/ChangeLog/20230827?id=notable-changes>, we wont realistically be fixing the broken user keymaps. Their removal in the next develop cycle would be highly likely.

Apteryks, 2023-09-14:
Removing them sounds reasonable, if nobody intends to fix them.

closed 2023-09-21 by waffle87

Apteryks, 2023-09-22:
Hi! I would perhaps leave this issue opened, as the fix in #22102 was fixing one out of the tens of failures. It may be handy to have this list around when pruning broken keymaps before a future release.