qmk/qmk_firmware #22050: Help w/ dfu-programmer failure -- dfu_download ERR: LIBUSB_ERROR_PIPE: Pipe error. 0xfffffff7 (-9)

alaricljs, 2023-09-16:
### Issue Description

Trying to load qmk onto a pair of sea-micro Atmega32U4 boards.  dfu-programmer is failing.

Short version:
```
sudo dfu-programmer atmega32u4 flash --force sofle_rev1_rgb_default.hex
0%                            100%  Programming 0x6500 bytes...
[ X  ERROR
Memory write error, use debug for more info.
```

Long version:
```
sudo dfu-programmer atmega32u4 flash --debug 9000--force sofle_rev1_rgb_default.hex
     target: atmega32u4
    chip_id: 0x2ff4
  vendor_id: 0x03eb
    command: flash
      quiet: false
      debug: 9000
device_type: AVR
------ command specific below ------
   validate: true
   hex file: sofle_rev1_rgb_default.hex

dfu.c:318: dfu_device_init( 1003, 12276, 0x7ffd6a5a2060, true, false )
dfu.c:322: dfu_device_init(000003eb, 00002ff4)
dfu.c:336:  0: 0x1d6b, 0x0002
dfu.c:336:  1: 0x1d6b, 0x0003
dfu.c:336:  2: 0x1d6b, 0x0003
dfu.c:336:  3: 0x1d6b, 0x0002
dfu.c:336:  4: 0x1d6b, 0x0003
dfu.c:336:  5: 0x1d6b, 0x0002
dfu.c:336:  6: 0x0bda, 0x0411
dfu.c:336:  7: 0x2109, 0x0812
dfu.c:336:  8: 0x2109, 0x0812
dfu.c:336:  9: 0x1d6b, 0x0003
dfu.c:336: 10: 0x03eb, 0x2ff4
dfu.c:346: found device at USB:3,29
dfu.c:502: dfu_find_interface()
dfu.c:512: config 0: MaxPower=50*2 mA
dfu.c:519: interface 0
dfu.c:526: setting 0: class:0, subclass 0, protocol:0
dfu.c:541: Found DFU Interface: 0
dfu.c:357: opened interface 0...
dfu.c:359: set configuration 1...
dfu.c:362: claimed interface 0...
dfu.c:292: dfu_abort( 0x7ffd6a5a2060 )
dfu.c:203: dfu_get_status( 0x7ffd6a5a2060, 0x7ffd6a5a2070 )
dfu.c:229: ==============================
dfu.c:230: status->bStatus: OK (0x00)
dfu.c:232: status->bwPollTimeout: 0x0001 ms
dfu.c:233: status->bState: dfuIDLE (0x02)
dfu.c:235: status->iString: 0x00
dfu.c:236: ------------------------------
dfu.c:569: State: dfuIDLE (2)
atmel.c:1151: atmel_flash( 0x7ffd6a5a2060, 0x7ffd6a5a1f50, false, false )
intel_hex.c:675: intel_flash_prep_buffer( 0x7ffd6a5a1f50 )
atmel.c:1189: Flash available from 0x0 to 0x6FFF (64kB p. 0 to 0), 0x7000 bytes.
atmel.c:1194: Data start @ 0x0: 64kB p 0; 128B p 0x0 + 0x0 offset.
atmel.c:1198: Data end @ 0x64FF: 64kB p 0; 128B p 0xC9 + 0x7F offset.
atmel.c:1202: Totals: 0x6500 bytes, 202 128B pages, 1 64kB byte pages.
atmel.c:819: atmel_blank_check( 0x7ffd6a5a2060, 0x00000000, 0x000064FF )
atmel.c:927: atmel_select_memory_unit( 0x7ffd6a5a2060, 0 )
atmel.c:941: Ignore Select Memory Unit for non GRP_AVR32 device.
Checking memory from 0x0 to 0x64FF...  
atmel.c:986: atmel_select_page( 0x7ffd6a5a2060, 0 )
atmel.c:999: Selecting page 0, address 0x0.
dfu.c:144: dfu_download( 0x7ffd6a5a2060, 4, 0x7ffd6a5a1e63 )
dfu.c:165: Message: m[0] = 0x06
dfu.c:165: Message: m[1] = 0x03
dfu.c:165: Message: m[2] = 0x00
dfu.c:165: Message: m[3] = 0x00
dfu.c:203: dfu_get_status( 0x7ffd6a5a2060, 0x7ffd6a5a1e54 )
dfu.c:229: ==============================
dfu.c:230: status->bStatus: OK (0x00)
dfu.c:232: status->bwPollTimeout: 0x0001 ms
dfu.c:233: status->bState: dfuIDLE (0x02)
dfu.c:235: status->iString: 0x00
dfu.c:236: ------------------------------
atmel.c:752: __atmel_blank_page_check( 0x7ffd6a5a2060, 0x00000000, 0x000064ff )
dfu.c:144: dfu_download( 0x7ffd6a5a2060, 6, 0x7ffd6a5a1ed0 )
dfu.c:165: Message: m[0] = 0x03
dfu.c:165: Message: m[1] = 0x01
dfu.c:165: Message: m[2] = 0x00
dfu.c:165: Message: m[3] = 0x00
dfu.c:165: Message: m[4] = 0x64
dfu.c:165: Message: m[5] = 0xff
dfu.c:203: dfu_get_status( 0x7ffd6a5a2060, 0x7ffd6a5a1ec0 )
dfu.c:229: ==============================
dfu.c:230: status->bStatus: OK (0x00)
dfu.c:232: status->bwPollTimeout: 0x0001 ms
dfu.c:233: status->bState: dfuIDLE (0x02)
dfu.c:235: status->iString: 0x00
dfu.c:236: ------------------------------
atmel.c:784: Flash region from 0x0 to 0x64FF is blank.
atmel.c:861: Flash blank from 0x0 to 0x64FF.
Empty.
atmel.c:927: atmel_select_memory_unit( 0x7ffd6a5a2060, 0 )
atmel.c:941: Ignore Select Memory Unit for non GRP_AVR32 device.
Programming 0x6500 bytes...
atmel.c:986: atmel_select_page( 0x7ffd6a5a2060, 0 )
atmel.c:999: Selecting page 0, address 0x0.
dfu.c:144: dfu_download( 0x7ffd6a5a2060, 4, 0x7ffd6a5a1ee3 )
dfu.c:165: Message: m[0] = 0x06
dfu.c:165: Message: m[1] = 0x03
dfu.c:165: Message: m[2] = 0x00
dfu.c:165: Message: m[3] = 0x00
dfu.c:203: dfu_get_status( 0x7ffd6a5a2060, 0x7ffd6a5a1ed4 )
dfu.c:229: ==============================
dfu.c:230: status->bStatus: OK (0x00)
dfu.c:232: status->bwPollTimeout: 0x0001 ms
dfu.c:233: status->bState: dfuIDLE (0x02)
dfu.c:235: status->iString: 0x00
dfu.c:236: ------------------------------
atmel.c:1286: Program data block: 0x0 to 0x3FF (p. 0), 0x400 bytes.
atmel.c:1423: __atmel_flash_block( 0x7ffd6a5a2060, 0x7ffd6a5a1f50, false )
atmel.c:1385: atmel_flash_populate_header( 0x7ffd6a5a1a40, 0x0, 0x3FF, false )
atmel.c:1339: atmel_flash_populate_footer( 0x7ffd6a5a1a40, 0x7ffd6a5a1e60, 65535, 65535, 65535 )
dfu.c:144: dfu_download( 0x7ffd6a5a2060, 1072, 0x7ffd6a5a1a40 )
dfu.c:165: Message: m[0] = 0x01
... OP: deleted, m[1] - m[1070] ...
dfu.c:165: Message: m[1071] = 0xff
dfu.c:694: dfu_download ERR: LIBUSB_ERROR_PIPE: Pipe error. 0xfffffff7 (-9)
atmel.c:1485: atmel_flash: flash data dfu_download failed.
atmel.c:1486: Expected message length of 1072, got -9.
atmel.c:1292: Error flashing the block: err -2.
ERROR
Memory write error, use debug for more info.
commands.c:529: Error writing memory data. (err -4)
```

QMK Doctor:
```
qmk doctor
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.2
Ψ QMK home: /home/lstrout/Code/qmk
Ψ Detected Linux (Arch Linux).
Ψ Git branch: master
Ψ Repo version: 0.22.3
Ψ - Latest master: 2023-09-15 16:39:34 -0700 (71d5d14c13) -- Rico Phoenix Project No. 1 Layout Additions (#22035)
Ψ - Latest upstream/master: 2023-09-15 16:39:34 -0700 (71d5d14c13) -- Rico Phoenix Project No. 1 Layout Additions (#22035)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2023-09-15 16:39:34 -0700 (71d5d14c13) -- Rico Phoenix Project No. 1 Layout Additions (#22035)
Ψ - Common ancestor with upstream/develop: None
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 13.2.0
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.2
Ψ Found dfu-programmer version 1.0.0
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-07-17 11:39:05 +0200 --  (da78eb37)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go
```

sigprof, 2023-09-16:
You should not need to use `sudo` if you have udev rules set up correctly (which you apparently have, because `qmk doctor` would complain otherwise, and Arch Linux is not one of special distros where the rules supplied by QMK don't work without modifications).

You need to run `dfu-programmer atmega32u4 erase --force` first, then immediately run `dfu-programmer atmega32u4 flash file.hex`.  The issue is that the `atmel-dfu` bootloader has an internal “erase done” flag which is set only after the actual erase operation (the blank flash check does not affect that flag); if that flag is not set, the bootloader just refuses to perform any read or write operations.  And you really need the `--force` options for `erase`, otherwise `dfu-programmer` performs a blank flash check and skips the erase if the flash is already blank (this may affect `qmk flash`, because it does not use the `--force` option for compatibility with some ancient versions of `dfu-programmer` which did not have that option).

alaricljs, 2023-09-16:
That did it, thanks so much.  qmk flash also went w/ catalina bootloader rather than dfu so that'll be the next thing I dig into.

closed 2023-09-16 by alaricljs