qmk/qmk_firmware #22067: [Bug] Media control (except volume up/down) cannot be used in Bluetooth mode in macOS

DragonsCgithub, 2023-09-19:
### Describe the Bug

When the keyboard was connected with Bluetooth mode, media controls such as "Media prev", "Media Next", and "Media play" (except vol alt +/-) will not work, but they all work well in wired mode.

I have ruled out Karabiner, as the problem occurs on my Windows computer as well.

I'm not sure whether it's a problem with the Bluetooth on qmk_firmware or with my setup. Now I have to use other keys and remap them in Karabiner to make sure my media controls work in Bluetooth mode. Any advice?

(I'm using Vial to configure my QMK setup, and since I'm not the compiler of the QMK firmware, I don't have a good view of the source code.)

### Keyboard Used

DK6064

### Operating System

macOS Ventura 13.4.1

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [X] Karabiner (macOS)

### Other keyboard-related software installed

BetterTouchTool

sigprof, 2023-09-19:
> DK6064

If this is something like https://www.aliexpress.com/item/1005005618274517.html, this board is not supported in the upstream QMK at the moment (and apparently not in vial-qmk either).  No idea what's actually in the firmware of that board (it might not even be the real QMK/VIA/Vial code — there are some boards that emulate just the VIA protocol, but use some completely different firmware implementation).

You could try to contact the seller and ask them to provide the firmware source (if the firmware is actually derived from QMK, legally they are obligated to provide the source according to the GNU GPL conditions, but these requirements often just get ignored).  Without the firmware source nothing could really be done (it might be possible to reverse engineer the PCB, but that could be hard, especially because this PCB also has a wireless module which probably communicates with the main MCU using some undocumented protocol).

However, even getting the firmware sources might not help in this case, because the wireless module (which runs its own firmware, probably some proprietary one) might just not have any support for sending those media keycodes over Bluetooth.

DragonsCgithub, 2023-09-21:
Thank you for your answer, I just want to know whether someone has encountered the same problem as me, but I can't register Aliexpress in Mainland China. For sure, having not the firmware source it was hard to know what was wrong (I had tried to contact the seller and ask them to provide the firmware source, but obviously I didn't receive it). Although the seller has replied that he will check the firmware, I don't have any hope (/-_-/). At present, I can only remap the media controls to some unused keys.

steamedham544, 2023-09-22:
I saw YMDK post a source to the firmware in their discord. They sell a version of the PCB without any wireless functionality. 

https://drive.google.com/file/d/1rzzfEwe6sSeuPXkg2KZfA34hlJP2JYzr/view

DragonsCgithub, 2023-09-23:
> I saw YMDK post a source to the firmware in their discord. They sell a version of the PCB without any wireless functionality.
> 
> https://drive.google.com/file/d/1rzzfEwe6sSeuPXkg2KZfA34hlJP2JYzr/view

ooh! Thank you very much for providing the firmware source. I am sure that the PCB sold by YMDK is similar to mine (even exactly the same), except that the PCB I purchased has an additional Bluetooth chip. I believe this source file can help me gain a deeper understanding of this keyboard.