qmk/qmk_firmware #22086: [Feature Request] Mod-tap but for normal key codes

boosh, 2023-09-20:
### Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [X] New behavior

### Description

I`d like to configure my keyboard to send 2 (different) normal key codes on tap or hold since I have RSI that`s aggravated by reaching my index finger. Specifically I'd like to be able to tap the 'r' key and have it output an 'r', but have it output a 't' on hold. Since this is to help my RSI I don't want to use tap dance for this.

It'd be good if there was a simple behaviour like mod-tap, but just for simple key codes.

zvecr, 2023-09-20:
Due to address space limitations, this cannot be achieved.

Closest you can get really is <https://docs.qmk.fm/#/mod_tap?id=intercepting-mod-taps>.

boosh, 2023-09-20:
Ah OK, I'll try that, thanks

closed 2023-09-20 by boosh