qmk/qmk_firmware #22092: Combining intercepting keys with leaders

boosh, 2023-09-20:
### Issue Description

I'm trying to do the following:

1. tap 'r' should output 'r'
2. hold 'r' should output 't'
3. Use 'r' as a leader, so 'r'->'f' outputs 'g'

The first 2 work, but I can't get the leader working. With the following, tapping r just prints a new line:
```
keymap = ... LT(0,QK_LEAD) ...

void leader_end_user(void) {
    if (leader_sequence_one_key(KC_F)) {
        // Leader, f => Types the below string
        tap_code16(KC_G);
    } else if (leader_sequence_timed_out()) {
        // default to the below
        tap_code16(KC_R);
    }
}

// Tap-hold for two normal key codes to shift left keys without holding a modifier
bool process_record_user(uint16_t keycode, keyrecord_t *record) {
    if (layer_state_cmp(layer_state | default_layer_state, QWERTY)) {
        switch (keycode) {
            case LT(0,QK_LEAD):
                if (!record->tap.count && record->event.pressed) {
                    tap_code16(KC_T); // Intercept hold function
                }
                return true;             // Normal processing of tap keycode handled in leader_end_user
...
```

Can these be used together?

sigprof, 2023-09-20:
`QK_LEAD` is not a basic keycode and cannot be used inside `LT()`, so you need to intercept both the hold and the tap functions of your custom keycode (the C equivalent of `QK_LEAD` is `leader_start()`).  You should probably use `LT(0, KC_R)` for your custom keycode, so that the behavior won't depend on the truncation of the `QK_LEAD` value to its low 8 bits.

Note that unless you define `LEADER_KEY_STRICT_KEY_PROCESSING`, the leader code would reduce any `LT()` or `MT()` keycodes to the inner basic keycode (and that happens irrespective of `record->tap.count`, so even if the timeout values would allow to use a hold inside a leader sequence, the result would be indistinguishable from a tap).

Also there is one trick which could effectively allow using multiple leader keys — if you do
```c
    leader_start();
    return true;
```
(instead of `return false;`), processing of the current keycode would continue and eventually would reach the leader code, and the keycode would be added to the leader sequence as its first element (with `LT()`/`MT()` stripped as described above; this would be another reason to use `LT(0, KC_R)` in your case).

Note that the leader code would consume all keys pressed within `LEADER_TIMEOUT`, even if those keys don't match any of the defined sequences; if you don't like that, you would need to add some more custom code (or maybe even implement the desired functionality completely in `process_record_user()` instead of using the core leader code).

boosh, 2023-09-20:
Yeah, steps 1 & 2 work with `LT(0, KC_R)`. It sounds tricky to implement this with leader code. I might do as you suggest and just do it all in `process_record_user`. The challenge is not outputting the leader key 'r' if it's part of a leader sequence, but outputting it otherwise. I guess I'll just maintain a buffer in my custom code and handle it there.

Thanks for the pointers.

closed 2023-09-20 by boosh