qmk/qmk_firmware #22110: [Bug]  Gmmk Numpad locks up after flash, locks up a few seconds after plugging in

jsznap, 2023-09-22:
### Describe the Bug

After flashing with via or default, after powering on numpad it is totally unresponsive after 3-5 seconds from powering. 

### Keyboard Used

qmk comopile -b gmmk/numpad -km via/default

### Operating System

mac OS

### qmk doctor Output

```text
❯ qmk doctor
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.2
Ψ QMK home: /Users/joshuasznapstajler/qmk_firmware
Ψ Detected macOS 14.0 (Intel).
Ψ Git branch: master
Ψ Repo version: 0.22.5
Ψ - Latest master: 2023-09-22 12:12:20 +1000 (abd432fd7a) -- Allow for `qmk compile -kb all`. (#22022)
Ψ - Latest upstream/master: 2023-09-22 12:12:20 +1000 (abd432fd7a) -- Allow for `qmk compile -kb all`. (#22022)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2023-09-22 12:12:20 +1000 (abd432fd7a) -- Allow for `qmk compile -kb all`. (#22022)
Ψ - Common ancestor with upstream/develop: None
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.5.0
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.2
Ψ Found dfu-programmer version 1.1.0
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-07-17 11:39:05 +0200 --  (da78eb37)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

jsznap, 2023-09-23:
New finding: The numpad with QMK flashed, works fine on windows but when connected to a mac, it freezes up and doesn't work after a few seconds.

pancaek, 2023-10-17:
I found the same issue with arch linux, but it works fine in my windows dual boot. Here's qmk doctor from my arch as well
```qmk doctor 
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.2
Ψ QMK home: /home/pancaek/qmk_firmware
Ψ Detected Linux (Arch Linux).
Ψ Git branch: dev_branch
Ψ Repo version: 0.21.6
Ψ - Latest dev_branch: 2023-10-16 20:24:57 -0700 (bafa47ec97) -- Merge branch 'qmk:master' into dev_branch
Ψ - Latest upstream/master: 2023-10-17 09:43:50 +1100 (f6c70c40af) -- Allow for disabling of parallel processing of qmk find and `qmk mass-compile`. (#22160)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2023-10-17 09:43:50 +1100 (f6c70c40af) -- Allow for disabling of parallel processing of qmk find and `qmk mass-compile`. (#22160)
Ψ - Common ancestor with upstream/develop: None
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 13.2.0
Ψ Found avr-gcc version 8.3.0
Ψ Found avrdude version 7.2
Ψ Found dfu-programmer version 1.0.0
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-07-17 11:39:05 +0200 --  (da78eb37)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go```

danremollino, 2023-12-31:
Also using macOS and my Numpad was locking up after about 5 seconds after being plugged in. I found that removing the code that handles the midi slider from keymap.c and re-compiling resolves the issue.

pancaek, 2024-01-02:
Interesting! I'll try this out as well