qmk/qmk_firmware #22167: c2json fails to find C file because the folder 'keyboards' is not added to the search path

akseidel, 2023-09-30:
### Describe the Bug

c2json logs the error 'C file does not exist!' (line 32) when using -kb and -km arguments that work just fine for compile. This is at an error check looking at cli.args.filename.exists().

Adding cli.log.error(cli.args.filename) prior to line 32 reveals all is good with the filename except that the folder 'keyboards' is missing. For example: qmk_firmware/adafruit/macropad/keymaps/mls001/keymap.c

### Keyboard Used

adafruit/macropad

### Operating System

OSX

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.2
Ψ QMK home: /Users/margaretbahe/qmk_firmware
Ψ Detected macOS 13.5.2 (Intel).
Ψ Git branch: master
Ψ Repo version: 0.21.7
Ψ - Latest master: 2023-08-26 19:43:40 +0100 (f07490bc09) -- Corrected duration example, should be number not string (#21839)
Ψ - Latest upstream/master: 2023-09-29 13:19:03 +0200 (cfb4bb9b98) -- teleport/native: switch from wrongly used user to kb function, boost matrix scan rate (#21172)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2023-08-26 19:43:40 +0100 (f07490bc09) -- Corrected duration example, should be number not string (#21839)
Ψ - Common ancestor with upstream/develop: None
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.5.0
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.2
Ψ Found dfu-programmer version 1.1.0
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-01-11 16:42:27 +0100 --  (a224be15)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

zvecr, 2023-09-30:
Can you post the actual invocation of `qmk c2json` please.

akseidel, 2023-10-01:
Ok, filename is not the same thing as filepathname. I get it to function when providing the filepathname less the qmk_firmware folder. Thank you.

Perhaps you can understand the confusion compared to qmk compile that evaluates all the required paths and the fact that the error checked cli.args.filename includes the /keymaps/ folder but omits the /keyboards/ folder. That certainly is strange. The missing element is actually present.

c2json --help could be more explicit. That would be the easiest fix.

Thank you

zvecr, 2023-10-01:
> compared to qmk compile that evaluates all the required paths

`qmk compile` doesnt have anything that would manipulate a provided filename to include additional locations outside of checking a relative path against the current working directory or the root of the `qmk_firmware` folder.

> and the fact that the error checked cli.args.filename includes the /keymaps/ folder but omits the /keyboards/ folder. That certainly is strange. The missing element is actually present.

Your issue was that the filename was missing 'keyboards/'? In which case the required missing element would definitely not have been present.

closed 2024-01-31 by github-actions[bot]