qmk/qmk_firmware #22171: [Bug] Windows Awaken from Sleep Automatically

samanxa, 2023-09-30:
### Describe the Bug

I used to put my computer to sleep at night. But this is not working when I am using qmk keyboards. My computer just awakes at night automatically. This is true not just for bioi/f60 but also keychron v2, seem to be a common pattern.

### Keyboard Used

bioi/f60

### Operating System

Windows

### qmk doctor Output

```text
Using web config tool
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

ephb, 2023-11-01:
I think you need to be a bit more specific here. Which keyboard are you using exactly? What was your build command for the firmware?
What was the last wake event recorded by windows?

I am using different QMK boards on multiple machines and none prevent sleep in Windows 10.

samanxa, 2023-11-03:
Keyboard is Percent Studio Trio 60 with board bioi/f60

It was built through qmk configurator.

100% sure that it doesn't wake up if I unplug qmk keyboard.

Last windows power events:
```
The system is exiting Modern Standby 

Reason: 16777220.

The system session has transitioned from 133 to 134.

Reason 16777220 

BootId: 41

The system is entering Modern Standby 

Reason: 16777220.

The system is exiting Modern Standby 

Reason: Input Mouse.

The system session has transitioned from 134 to 136.

Reason InputHid 

BootId: 41
```

ephb, 2023-11-05:
Thank you for providing the details.

Out of these 5 events only two may be related to your keyboard.

According to Microsoft [documentation](https://learn.microsoft.com/en-us/windows-hardware/design/device-experiences/modern-standby-sleepstudy) the other 3 are scheduled tasks.

If you have built your keyboard with mouse keys, then the mouse event could be your keyboard and only the last one, with HID in its description, likely is your keyboard.

drashna, 2023-11-10:
For windows, "powercfg -lastwake" will give you the device or driver that triggered the last wakeup event.  This is generally better for checking where the issue is. 

Also, that you have "input mouse" listed here, and that the keyboard in question doesn't appear to support pointing devices, this is likely due to your mouse, not the keyboard.