qmk/qmk_firmware #22190: [Feature Request] Consider dropping support for Python 3.6(7)

tomasr8, 2023-10-02:
### Feature Request Type

- [X] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

As of June of this year, Python 3.7 no [longer receives security patches](https://endoflife.date/python). We should consider bumping the minimum python version to at least 3.8 (but realistically even 3.9 should be fine at this point).

As an aside, it's not clear to me whether the current minimum version is 3.6 or 3.7. [The docs say](https://github.com/qmk/qmk_firmware/blob/master/docs/cli.md) 3.6:

> QMK requires Python 3.6 or greater

But the [website ](https://docs.qmk.fm/#/coding_conventions_python?id=testing)states 3.7:

> We target Python 3.7 for compatibility with all supported platforms.

Is it 3.6 or 3.7? 

(Btw, thanks a lot for this great piece of software! All my keyboards run on QMK :))

zvecr, 2023-10-02:
Minimum is currently 3.7, as enforced by:

https://github.com/qmk/qmk_cli/blob/82bb65bd5236c3247dbc4db5d82beb25f4c9fd38/qmk_cli/script_qmk.py#L49-L51

The 3.6 references in the docs are incorrect, and were missed last version bump.

tomasr8, 2023-10-03:
I see.. If we really want to support all LTS versions, we'll have to stick with 3.7 for a while since it's the default on Debian 10.

PR #24554 (2024-10-31): Bump minimum required Python version

## Description

Updates to reflect the `1.1.6` release of `qmk_cli`.

## Types of Changes

- [ ] Core
- [ ] Bugfix
- [ ] New feature
- [x] Enhancement/optimization
- [ ] Keyboard (addition or update)
- [ ] Keymap/layout (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Fixes #22190

## Checklist

- [x] My code follows the code style of this project: [**C**](https://docs.qmk.fm/#/coding_conventions_c), [**Python**](https://docs.qmk.fm/#/coding_conventions_python)
- [x] I have read the [**PR Checklist** document](https://docs. [... cut]
closed 2024-02-02 by github-actions[bot]