qmk/qmk_firmware #22206: [Feature Request] Locking modifier

skagon, 2023-10-05:
### Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [X] Alteration (enhancement/optimization) of existing feature(s)
- [X] New behavior

### Description

I wonder if it'd be possible to mark one specific key – which would have to be a modifier otherwise it makes no sense – to operate as a locking key. I mean, exactly like Caps Lock works: press once, it's on, press a second time, it's off. Only I want to do it for LEFT_ALT. Is there an option I haven't noticed? I just want to designate a key (in a specific layer) to act as a locking left Alt. Press once, LALT remains "pressed", I press more keys, then press *key* again, LALT gets released. Of course, you can do it with the KEY_LOCK_ENABLE and QK_LOCK combination, but it's a nuisance to have to press one extra key. Is there an option to integrate QK_LOCK functionality in one key?

fauxpark, 2023-10-05:
Caps Lock's functionality is not handled by the keyboard, it works the same as any other key and the OS handles the "locking" part.

But you could emulate this by making a custom keycode that keeps track of the state:
```c
bool alt_lock_state = false;

bool process_record_user(uint16_t keycode, keyrecord_t *record) {
    switch (keycode) {
        case ALT_LOCK:
            if (record->event.pressed) {
                alt_lock_state = !alt_lock_state;

                if (alt_lock_state) {
                    register_code(KC_LALT);
                } else {
                    unregister_code(KC_LALT);
                }
            }
            return false;
    }

    return true;
}
```

skagon, 2023-10-05:
> Caps Lock's functionality is not handled by the keyboard, it works the same as any other key and the OS handles the "locking" part.
> 
> But you could emulate this by making a custom keycode that keeps track of the state:
> 

This is helpful, thanks. I wonder if there is any interest in this functionality, so that it is made a permanent QK keycode / modifier.
However, I've made a slight tweak, just… because OCD.

```
bool alt_lock_state = false;

bool process_record_user(uint16_t keycode, keyrecord_t *record)
{
    switch (keycode)
	{
        case ALT_LOCK:
            if (record->event.pressed && !alt_lock_state)
			{
				register_code(KC_ALT);
				alt_lock_state = !alt_lock_state;
			}
			else if (record->event.released && alt_lock_state)
			{
				unregister_code(KC_ALT);
                alt_lock_state = !alt_lock_state;
            }
            return false;
    }

    return true;
}
```
I'm off to try it out. Thanks again!

closed 2024-02-04 by github-actions[bot]