qmk/qmk_firmware #22208: [Bug] DUMMY_MOD_NEUTRALIZER doesn't work as intended

iQuerz, 2023-10-05:
### Describe the Bug

The dummy mod neutralizer feature in key overrides feature is not being called after the suppression restores the modifier key.

Example: I have this bit of code that should let me use alt+9 for `KC_HOME` in `keymap.c`:
```cpp
const key_override_t alt_9_override = ko_make_basic(MOD_BIT(KC_LEFT_ALT), KC_9, KC_HOME);
const key_override_t **key_overrides = (const key_override_t *[]){
    &alt_9_override,
    NULL
};
```
with `config.h`:
```cpp
#define DUMMY_MOD_NEUTRALIZER_KEYCODE KC_RIGHT_CTRL
#define MODS_TO_NEUTRALIZE { MOD_BIT(KC_LEFT_ALT) }
```
but when I tested my alt+9 on keytester, this is the output I got:
[image]

When the suppression reactivates the alt button, dummy neutralizer should be fired as well. If left like this, my alt+9 fires `KC_HOME` + `KC_LALT`, instead of just `KC_HOME`

### Keyboard Used

kprepublic/bm60hsrgb_poker/rev2

### Link to product page (if applicable)

https://kprepublic.com/products/bm60-rgb-60-gh60-hot-swappable-pcb-programmed-qmk-firmware-type-c

### Operating System

Windows

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: C:/Users/Administrator/qmk_firmware
Ψ Detected Windows 10 (10.0.19045).
Ψ QMK MSYS version: 1.7.2
Ψ Git branch: master
Ψ Repo version: 0.22.10
⚠ Git has unstashed/uncommitted changes.
Ψ - Latest master: 2023-10-04 03:17:01 -0700 (6e93b59f61) -- Monstargear XO87 Soldered: correct layout data (#22199)
Ψ - Latest upstream/master: 2023-10-04 21:55:49 -0700 (92873cc14f) -- Add printedpad keyboard (#21202)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2023-10-04 03:17:01 -0700 (6e93b59f61) -- Monstargear XO87 Soldered: correct layout data (#22199)
Ψ - Common ancestor with upstream/develop: None
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.1.0
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.0
Ψ Found dfu-programmer version 0.7.2
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-07-17 11:39:05 +0200 --  (da78eb37)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [X] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Additional Context

I have AutoHotkey installed but my goal is to get rid of it, that's why I'm making these overrides in QMK. All testing was done with supressed hotkeys in autohotkey.

iQuerz, 2023-10-05:
@precondition this is the issue I've mentioned on the discord server. Thank you for looking into it :)

precondition, 2023-10-05:
As far as I know, the code of the key overrides feature is not called when the modifier is released so I suspect that patching this issue will require turning on a flag in `quantum/process_keycode/process_key_override.c` and checking for the activation of that flag in the core key processing code (e.g. `if key.is_mod() && !key.pressed && MOD_BIT(key) & MODS_TO_NEUTRALIZE && mod_was_neutralized) { tap_code(DUMMY_MOD_CODE_NEUTRALIZER); } unregister_code(key);`)

sigprof, 2023-10-05:
One option here is to emit the neutralizer in `clear_active_override()` if we are reregistering an affected modifier which was suppressed.

drewlwhitney, 2024-06-19:
I am also interested in this! I'm attempting to use GUI with a keycode override, and the DUMMY_MOD_NEUTRALIZER still does not work. I found a workaround by using process_record_user() and sending separate keyup and keydown events, but it seems a bit impractical.