qmk/qmk_firmware #22275: [Bug] Piantor v2 3x5 layout issue

sondrelg, 2023-10-14:
### Describe the Bug

When flashing this simple layout

[image]

My button for `a` outputs `s`, and my button for `g` outputs nothing. It seems that the whole row is shifted over one to the right.

I'm completely new to both qmk and the piantor keyboard, so not sure if this is a software or hardware problem. Any ideas? 

### Keyboard Used

beekeeb/piantor

### Operating System

Pop!_OS 22.04

### qmk doctor Output

```text
❯ qmk doctor                            
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.2
Ψ QMK home: /home/sondrelg/qmk_firmware
Ψ Detected Linux (Pop!_OS 22.04 LTS).
⚠ Missing or outdated udev rules for 'atmel-dfu' boards. Run 'sudo cp /home/sondrelg/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'kiibohd' boards. Run 'sudo cp /home/sondrelg/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'stm32-dfu' boards. Run 'sudo cp /home/sondrelg/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'apm32-dfu' boards. Run 'sudo cp /home/sondrelg/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'gd32v-dfu' boards. Run 'sudo cp /home/sondrelg/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'wb32-dfu' boards. Run 'sudo cp /home/sondrelg/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'bootloadhid' boards. Run 'sudo cp /home/sondrelg/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'usbasploader' boards. Run 'sudo cp /home/sondrelg/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'usbtinyisp' boards. Run 'sudo cp /home/sondrelg/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'md-boot' boards. Run 'sudo cp /home/sondrelg/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Detected ModemManager without the necessary udev rules. Please either disable it or set the appropriate udev rules if you are using a Pro Micro.
⚠ Missing or outdated udev rules for 'caterina' boards. Run 'sudo cp /home/sondrelg/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'hid-bootloader' boards. Run 'sudo cp /home/sondrelg/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
Ψ Git branch: master
Ψ Repo version: 0.22.12
Ψ - Latest master: 2023-10-14 23:10:55 +1100 (81a3aa025c) -- Fix typo in keychron v6 encoder map enable (#22272)
Ψ - Latest upstream/master: 2023-10-14 23:10:55 +1100 (81a3aa025c) -- Fix typo in keychron v6 encoder map enable (#22272)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2023-10-14 23:10:55 +1100 (81a3aa025c) -- Fix typo in keychron v6 encoder map enable (#22272)
Ψ - Common ancestor with upstream/develop: None
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.3.1
Ψ Found avr-gcc version 5.4.0
Ψ Found avrdude version 6.3-20171130
Ψ Found dfu-programmer version 0.6.1
Ψ Found dfu-util version 0.9
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-07-17 11:39:05 +0200 --  (da78eb37)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go, but minor problems were found
```
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

sondrelg, 2023-10-15:
Thanks for the quick response @lesshonor. Agree, seems likely. I've sent Leo from beekeeb an email, and referenced this issue :slightly_smiling_face: I suppose it might be nice to resolve it here if possible, so other users can find the issue?

MTW2237-40, 2023-10-22:
The firmware on my Piantor v2 3x5 has a scrambled keymap in no seeming order, and when I try to flash my it, it just scrambles again. How could this be solved?

MTW2237-40, 2023-10-22:
This is the output from qmk doctor.
`Ψ CLI version: 1.1.1
Ψ QMK home: C:/Users/chant/qmk_firmware
Ψ Detected Windows 10 (10.0.22621).
Ψ QMK MSYS version: 1.7.2
Ψ Git branch: master
Ψ Repo version: 0.22.12
Ψ - Latest master: 2023-10-17 09:43:50 +1100 (f6c70c40af) -- Allow for disabling of parallel processing of qmk find and `qmk mass-compile`. (#22160)
Ψ - Latest upstream/master: 2023-10-21 21:31:23 +1100 (0e6d47e1e4) -- Fixup symbol sizing listing. (#22307)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2023-10-17 09:43:50 +1100 (f6c70c40af) -- Allow for disabling of parallel processing of qmk find and `qmk mass-compile`. (#22160)
Ψ - Common ancestor with upstream/develop: None
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.1.0
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.0
Ψ Found dfu-programmer version 0.7.2
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-07-17 11:39:05 +0200 --  (da78eb37)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go`

ChocolateOverflow, 2023-11-04:
I also had an issue using `LAYOUT_split_3x5_3` as the columns got shifted by 1. I fixed/hacked it by using `LAYOUT_split_3x6_3` and padding the 2 outer columns with `XXXXXXX` keys (imagine they died when Leo snapped the column off, but the phantom ~~limbs~~ keys are still there in software).

closed 2024-01-08 by drashna