qmk/qmk_firmware #22291: [Bug] Can't compile lily58 keyboard for rp2040

jittapont, 2023-10-18:
### Describe the Bug

I tried to run command to compile the u2f files for my lily58 keyboard `qmk compile -kb lily58/rev1 -km via -e CONVERT_TO=promicro_rp2040` but got the compilation errors
```
Copyright (C) 2023 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: .build/obj_lily58_rev1_via/src/default_keyboard.c                                       In file included from ./lib/pico-sdk/src/common/pico_base/include/pico/types.h:12,
                 from ./lib/pico-sdk/src/common/pico_base/include/pico.h:24,
                 from ./lib/pico-sdk/src/rp2_common/hardware_flash/include/hardware/flash.h:10,
                 from ./platforms/chibios/drivers/wear_leveling/wear_leveling_rp2040_flash_config.h:6,
                 from <command-line>:
./lib/pico-sdk/src/common/pico_base/include/pico/assert.h:18:10: fatal error: assert.h: No such file or directory
   18 | #include <assert.h>
      |          ^~~~~~~~~~
compilation terminated.
 [ERRORS]
 |
 |
 |
gmake[1]: *** [builddefs/common_rules.mk:361: .build/obj_lily58_rev1_via/.build/obj_lily58_rev1_via/src/default_keyboard.o] Error 1
gmake: *** [Makefile:392: lily58/rev1:via] Error 1
Make finished with errors
```

### Keyboard Used

lily58/rev1

### Operating System

MacOS

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.2
Ψ QMK home: /Users/jittapon/qmk_firmware
Ψ Detected macOS 11.4 (Intel).
Ψ Git branch: master
Ψ Repo version: 0.22.12
⚠ Git has unstashed/uncommitted changes.
Ψ - Latest master: 2023-10-17 09:43:50 +1100 (f6c70c40af) -- Allow for disabling of parallel processing of qmk find and `qmk mass-compile`. (#22160)
Ψ - Latest upstream/master: 2023-10-17 09:43:50 +1100 (f6c70c40af) -- Allow for disabling of parallel processing of qmk find and `qmk mass-compile`. (#22160)
Ψ - Latest upstream/develop: 2023-10-16 22:44:27 +0000 (20cefe254d) -- Merge remote-tracking branch 'origin/master' into develop
Ψ - Common ancestor with upstream/master: 2023-10-17 09:43:50 +1100 (f6c70c40af) -- Allow for disabling of parallel processing of qmk find and `qmk mass-compile`. (#22160)
Ψ - Common ancestor with upstream/develop: 2023-10-17 09:43:50 +1100 (f6c70c40af) -- Allow for disabling of parallel processing of qmk find and `qmk mass-compile`. (#22160)
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 13.2.0
Ψ Found avr-gcc version 9.4.0
⚠ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.
Ψ Found avrdude version 7.2
Ψ Found dfu-programmer version 1.1.0
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-07-17 11:39:05 +0200 --  (da78eb37)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [X] Karabiner (macOS)

elpekenin, 2023-10-18:
This is apparently due to a missing/incomplete installation of the compiler toolchain, a user reported this on discord as getting fixed after installing "arm-none-eabi-newlib" package. Beware, name could be different on other package manager/distros

fauxpark, 2023-10-18:
Above is incorrect, you have the *wrong* arm-none-eabi toolchain. Please uninstall it and reinstall `qmk/qmk/qmk` in Homebrew.

jittapont, 2023-10-22:
Thanks a lot 🙇

closed 2023-10-22 by fauxpark

malparty, 2024-03-08:
Adding a note in case a simple `uninstall / reinstall` do not work for you:

> `brew uninstall qmk/qmk/qmk`, then `brew uninstall arm-none-eabi-gcc`. 
> Then `brew install qmk/qmk/qmk` flow again

Source: https://www.reddit.com/r/olkb/comments/179w4us/need_some_help_about_compiling_the_lily58/

wwalker, 2024-08-15:
I had this problem on Fedora 40 (brew's version of qmk/qmk/qmk doesn't install on Linux of any flavor).

The fix on Fedora 40 (and probably all linux flavors) is:

```
qmk setup # this install avr-none-eabi-gcc, but not newlib
util/qmk_install.sh # This will install avr-none-eabi-newlib , which, after make clean, fixes it
make clean
qmk compile -kb dasky/reverb -km default
```

fauxpark, 2024-08-15:
That won't fix anything. `qmk setup` does run `util/qmk_install.sh`. Something else is wrong with your setup here.

wwalker, 2024-08-15:
fauxpark, It is highly repeatable.  Here is a text file showing that `qmk setup`, while it may run `util/qmk_install.sh` under the covers, it does not install `arm-none-eabi-newlib`, which is critical.

[highly-repeatable.txt.gz](https://github.com/user-attachments/files/16624494/highly-repeatable.txt.gz)

I also ran :

```
strace -f -s 256 -o ~/tmp/what-in-the-world -tt qmk setup
```

while `qmk setup` does call `fstat()` on `util/qmk_install.sh`, it never `exec()`s, nor `open()`s it, even though the necessary `arm-none-eabi-newlib` is not installed when qmk_setup is run.

```
@$ grep qmk_install ~/tmp/what-in-the-world
3919707 06:08:21.769036 newfstatat(AT_FDCWD, "util/qmk_install.sh",  <unfinished ...>
@$ grep -A20 qmk_install ~/tmp/what-in-the-world | grep 3919707
3919707 06:08:21.769036 newfstatat(AT_FDCWD, "util/qmk_install.sh",  <unfinished ...>
3919707 06:08:21.769103 <... newfstatat resumed>{st_mode=S_IFREG|0755, st_size=2847, ...}, AT_SYMLINK_NOFOLLOW) = 0
```