qmk/qmk_firmware #22300: Not able to compile

Harrycheng1992, 2023-10-20:
### Issue Description

I'm currently working on a split keyboard project where I'm using I2C to connect two keyboard halves. In this post, I'd like to share the relevant files and seek some advice on my implementation.

I have attached the files for the "master_left" side of the keyboard.

I'm encountering an issue while compiling the firmware for my 8x11 split keyboard(left). A few weeks ago, I was able to successfully compile the firmware and use the keyboard without any problems. However, now I'm facing an error when running the command qmk compile -kb splitkeyboardv5d1 -km default.

I am hoping for some help to resolve a problem I've encountered.

The error message is provided below:

`Compiling: quantum/matrix.c                                                                        In file included from platforms/pin_defs.h:22,
                 from platforms/gpio.h:18,
                 from quantum/matrix.h:22,
                 from quantum/matrix.c:21:
platforms/avr/_pin_defs.h:58:27: error: excess elements in array initializer [-Werror]
 #define PINDEF(port, pin) ((PIN##port##_ADDRESS << PORT_SHIFTER) | pin)
                           ^
platforms/avr/_pin_defs.h:102:16: note: in expansion of macro 'PINDEF'
 #    define D2 PINDEF(D, 2)
                ^~~~~~
./keyboards/splitkeyboardv5d1/config.h:29:43: note: in expansion of macro 'D2'
 #define MATRIX_ROW_PINS { B0, B1, B2, B3, D2, D3, C6, C7 }
                                           ^~
quantum/matrix.c:58:58: note: in expansion of macro 'MATRIX_ROW_PINS'
 static SPLIT_MUTABLE_ROW pin_t row_pins[ROWS_PER_HAND] = MATRIX_ROW_PINS;
                                                          ^~~~~~~~~~~~~~~
platforms/avr/_pin_defs.h:58:27: note: (near initialization for 'row_pins')
 #define PINDEF(port, pin) ((PIN##port##_ADDRESS << PORT_SHIFTER) | pin)
                           ^
platforms/avr/_pin_defs.h:102:16: note: in expansion of macro 'PINDEF'
 #    define D2 PINDEF(D, 2)
                ^~~~~~
./keyboards/splitkeyboardv5d1/config.h:29:43: note: in expansion of macro 'D2'
 #define MATRIX_ROW_PINS { B0, B1, B2, B3, D2, D3, C6, C7 }
                                           ^~
quantum/matrix.c:58:58: note: in expansion of macro 'MATRIX_ROW_PINS'
 static SPLIT_MUTABLE_ROW pin_t row_pins[ROWS_PER_HAND] = MATRIX_ROW_PINS;
                                                          ^~~~~~~~~~~~~~~
platforms/avr/_pin_defs.h:58:27: error: excess elements in array initializer [-Werror]
 #define PINDEF(port, pin) ((PIN##port##_ADDRESS << PORT_SHIFTER) | pin)
                           ^
platforms/avr/_pin_defs.h:103:16: note: in expansion of macro 'PINDEF'
 #    define D3 PINDEF(D, 3)
                ^~~~~~
./keyboards/splitkeyboardv5d1/config.h:29:47: note: in expansion of macro 'D3'
 #define MATRIX_ROW_PINS { B0, B1, B2, B3, D2, D3, C6, C7 }
                                               ^~
quantum/matrix.c:58:58: note: in expansion of macro 'MATRIX_ROW_PINS'
 static SPLIT_MUTABLE_ROW pin_t row_pins[ROWS_PER_HAND] = MATRIX_ROW_PINS;
                                                          ^~~~~~~~~~~~~~~
platforms/avr/_pin_defs.h:58:27: note: (near initialization for 'row_pins')
 #define PINDEF(port, pin) ((PIN##port##_ADDRESS << PORT_SHIFTER) | pin)
                           ^
platforms/avr/_pin_defs.h:103:16: note: in expansion of macro 'PINDEF'
 #    define D3 PINDEF(D, 3)
                ^~~~~~
./keyboards/splitkeyboardv5d1/config.h:29:47: note: in expansion of macro 'D3'
 #define MATRIX_ROW_PINS { B0, B1, B2, B3, D2, D3, C6, C7 }
                                               ^~
quantum/matrix.c:58:58: note: in expansion of macro 'MATRIX_ROW_PINS'
 static SPLIT_MUTABLE_ROW pin_t row_pins[ROWS_PER_HAND] = MATRIX_ROW_PINS;
                                                          ^~~~~~~~~~~~~~~
platforms/avr/_pin_defs.h:58:27: error: excess elements in array initializer [-Werror]
 #define PINDEF(port, pin) ((PIN##port##_ADDRESS << PORT_SHIFTER) | pin)
                           ^
platforms/avr/_pin_defs.h:96:16: note: in expansion of macro 'PINDEF'
 #    define C6 PINDEF(C, 6)
                ^~~~~~
[... 177 lines cut ...]
/*communication option*/
#define USE_I2C

#define FORCED_SYNC_THROTTLE_MS 100

//This makes it possible to use a master part without the slave part connected.
#define SPLIT_MAX_CONNECTION_ERRORS 10
`  
keymap.c : 
`#include QMK_KEYBOARD_H

const uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {
   
    [0] = LAYOUT(
       	KC_ESC, KC_F1, KC_F2, KC_F3, KC_F4, KC_F5, KC_F6, 
		KC_GRV, KC_1, KC_2, KC_3, KC_4, KC_5, KC_MINS,
		KC_TAB, KC_Q, KC_W, KC_E, KC_R, KC_T, KC_LBRC,
		KC_CAPS, KC_A, KC_S, KC_D, KC_F, KC_G, KC_QUOT, 
		KC_LSFT, KC_Z, KC_X, KC_C, KC_V, KC_B,
		KC_LCTL, KC_LEFT_GUI, KC_LALT, KC_APP, KC_SPC, KC_ENT, KC_PGUP,
		KC_SPC, KC_PGDN,
		KC_SPC, KC_INS
   )
};
`

fauxpark, 2023-10-20:
Please make sure your repo is up to date, then run `qmk new-keyboard`.

closed 2023-10-20 by fauxpark

Harrycheng1992, 2023-10-20:
> Please make sure your repo is up to date, then run `qmk new-keyboard`.

You mean `git pull origin master` then new a keyboard?

fauxpark, 2023-10-20:
Your code is outdated. You need to start over with the proper boilerplate.

Harrycheng1992, 2023-10-20:
> Your code is outdated. You need to start over with the proper boilerplate.

Thank you, I was able to compile after rewriting the code.