qmk/qmk_firmware #22302: Working on a Split Keyboard with compiling issued

Harrycheng1992, 2023-10-20:
### Issue Description

I am currently in the process of creating a split keyboard that utilizes I2C to connect both sides. I'm encountering some issues, and I'd like to share the relevant files and describe the problem I'm facing.

I'm currently a beginner in making custom keyboards and I'm not sure if both sides of the keyboard need to flash the same hex file. I kindly request assistance from the community. Your help is greatly appreciated

The error : 
```
Compiling: quantum/encoder.c                                                                       quantum/encoder.c:62:48: error: excess elements in array initializer [-Werror]
 static uint8_t encoder_state[NUM_ENCODERS]  = {0};
                                                ^
quantum/encoder.c:62:48: note: (near initialization for 'encoder_state')
quantum/encoder.c:63:48: error: excess elements in array initializer [-Werror]
 static int8_t  encoder_pulses[NUM_ENCODERS] = {0};
                                                ^
quantum/encoder.c:63:48: note: (near initialization for 'encoder_pulses')
quantum/encoder.c:74:47: error: excess elements in array initializer [-Werror]
 static uint8_t encoder_value[NUM_ENCODERS] = {0};
                                               ^
quantum/encoder.c:74:47: note: (near initialization for 'encoder_value')
cc1: all warnings being treated as errors
 [ERRORS]
 | 
 | 
 | 
gmake[1]: *** [builddefs/common_rules.mk:361: .build/obj_20231020right_default/quantum/encoder.o] Error 1
gmake: *** [Makefile:392: 20231020right:default] Error 1
Make finished with errors
```

The following are my files details:

info.json:
```
{
    "manufacturer": "Harrycheng1992",
    "keyboard_name": "20231020right",
    "maintainer": "Harrycheng1992",
    "bootloader": "caterina",
    "diode_direction": "COL2ROW",
    "features": {
        "bootmagic": true,
        "command": false,
        "console": false,
        "extrakey": true,
        "mousekey": true,
        "nkro": true
    },
    "matrix_pins": {
        "cols": ["F0", "F1", "F4", "F5", "F6", "F7", "B4", "B5", "B6", "D7", "D6"],
        "rows": ["B0", "B1", "B2", "B3", "D2", "D3", "C6", "C7"]
    },
    "processor": "atmega32u4",
    "url": "",
    "usb": {
        "device_version": "1.0.0",
        "pid": "0x0000",
        "vid": "0xFEED"
    },
    "layouts": {
        "LAYOUT": {
            "layout": [
                {"matrix": [0, 0], "x": 0, "y": 0},
                {"matrix": [0, 2], "x": 2, "y": 0},
                {"matrix": [0, 3], "x": 3, "y": 0},
                {"matrix": [0, 4], "x": 4, "y": 0},
                {"matrix": [0, 5], "x": 5, "y": 0},
                {"matrix": [0, 6], "x": 6, "y": 0},
                {"matrix": [0, 7], "x": 7, "y": 0},
                {"matrix": [1, 2], "x": 2, "y": 1},
                {"matrix": [1, 3], "x": 3, "y": 1},
                {"matrix": [1, 4], "x": 4, "y": 1},
                {"matrix": [1, 5], "x": 5, "y": 1},
                {"matrix": [1, 6], "x": 6, "y": 1},
                {"matrix": [1, 7], "x": 7, "y": 1},
                {"matrix": [1, 8], "x": 8, "y": 1},
                {"matrix": [2, 0], "x": 0, "y": 2},
                {"matrix": [2, 3], "x": 3, "y": 2},
                {"matrix": [2, 4], "x": 4, "y": 2},
                {"matrix": [2, 5], "x": 5, "y": 2},
                {"matrix": [2, 6], "x": 6, "y": 2},
                {"matrix": [2, 7], "x": 7, "y": 2},
                {"matrix": [2, 8], "x": 8, "y": 2},
                {"matrix": [3, 0], "x": 0, "y": 3},
                {"matrix": [3, 3], "x": 3, "y": 3},
                {"matrix": [3, 4], "x": 4, "y": 3},
                {"matrix": [3, 5], "x": 5, "y": 3},
                {"matrix": [3, 6], "x": 6, "y": 3},
                {"matrix": [3, 7], "x": 7, "y": 3},
                {"matrix": [3, 8], "x": 8, "y": 3},
                {"matrix": [4, 1], "x": 1, "y": 4},
                {"matrix": [4, 3], "x": 3, "y": 4},
                {"matrix": [4, 4], "x": 4, "y": 4},
                {"matrix": [4, 5], "x": 5, "y": 4},
                {"matrix": [4, 6], "x": 6, "y": 4},
                {"matrix": [4, 7], "x": 7, "y": 4},
                {"matrix": [5, 0], "x": 0, "y": 5},
                {"matrix": [5, 1], "x": 1, "y": 5},
                {"matrix": [5, 3], "x": 3, "y": 5},
                {"matrix": [5, 4], "x": 4, "y": 5},
                {"matrix": [5, 6], "x": 6, "y": 5},
                {"matrix": [5, 9], "x": 9, "y": 5},
                {"matrix": [5, 10], "x": 10, "y": 5},
                {"matrix": [6, 7], "x": 7, "y": 6},
                {"matrix": [6, 10], "x": 10, "y": 6},
                {"matrix": [7, 8], "x": 8, "y": 7},
                {"matrix": [7, 9], "x": 9, "y": 7},
                {"matrix": [8, 3], "x": 3, "y": 8},
                {"matrix": [8, 4], "x": 4, "y": 8},
                {"matrix": [8, 5], "x": 5, "y": 8},
                {"matrix": [8, 6], "x": 6, "y": 8},
                {"matrix": [8, 7], "x": 7, "y": 8},
                {"matrix": [8, 8], "x": 8, "y": 8},
                {"matrix": [8, 10], "x": 10, "y": 8},
                {"matrix": [9, 2], "x": 2, "y": 9},
                {"matrix": [9, 3], "x": 3, "y": 9},
                {"matrix": [9, 4], "x": 4, "y": 9},
                {"matrix": [9, 5], "x": 5, "y": 9},
                {"matrix": [9, 6], "x": 6, "y": 9},
                {"matrix": [9, 7], "x": 7, "y": 9},
                {"matrix": [9, 9], "x": 9, "y": 9},
                {"matrix": [10, 2], "x": 2, "y": 10},
                {"matrix": [10, 3], "x": 3, "y": 10},
                {"matrix": [10, 4], "x": 4, "y": 10},
                {"matrix": [10, 5], "x": 5, "y": 10},
                {"matrix": [10, 6], "x": 6, "y": 10},
                {"matrix": [10, 7], "x": 7, "y": 10},
                {"matrix": [10, 9], "x": 9, "y": 10},
                {"matrix": [11, 2], "x": 2, "y": 11},
                {"matrix": [11, 3], "x": 3, "y": 11},
                {"matrix": [11, 4], "x": 4, "y": 11},
                {"matrix": [11, 5], "x": 5, "y": 11},
                {"matrix": [11, 6], "x": 6, "y": 11},
                {"matrix": [11, 7], "x": 7, "y": 11},
                {"matrix": [11, 9], "x": 9, "y": 11},
                {"matrix": [12, 3], "x": 3, "y": 12},
                {"matrix": [12, 4], "x": 4, "y": 12},
                {"matrix": [12, 5], "x": 5, "y": 12},
                {"matrix": [12, 6], "x": 6, "y": 12},
                {"matrix": [12, 7], "x": 7, "y": 12},
                {"matrix": [12, 9], "x": 9, "y": 12},
                {"matrix": [13, 0], "x": 0, "y": 13},
                {"matrix": [13, 1], "x": 1, "y": 13},
                {"matrix": [14, 1], "x": 1, "y": 14},
                {"matrix": [14, 4], "x": 4, "y": 14},
                {"matrix": [14, 6], "x": 6, "y": 14},
                {"matrix": [14, 7], "x": 7, "y": 14},
                {"matrix": [14, 8], "x": 8, "y": 14},
                {"matrix": [14, 10], "x": 10, "y": 14},
                {"matrix": [15, 0], "x": 0, "y": 15},
                {"matrix": [15, 1], "x": 1, "y": 15},
                {"matrix": [15, 3], "x": 3, "y": 15}
            ]
        }
    }
}
```

config.h:
```
#pragma once

#define LOCKING_SUPPORT_ENABLE
#define LOCKING_RESYNC_ENABLE
```

keycap.c:
```
#include QMK_KEYBOARD_H

const uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {

    [0] = LAYOUT(
        KC_ESC, KC_F1, KC_F2, KC_F3, KC_F4, KC_F5, KC_F6, 
		KC_GRV, KC_1, KC_2, KC_3, KC_4, KC_5, KC_MINS,
		KC_TAB, KC_Q, KC_W, KC_E, KC_R, KC_T, KC_LBRC,
		KC_CAPS, KC_A, KC_S, KC_D, KC_F, KC_G, KC_QUOT, 
		KC_LSFT, KC_Z, KC_X, KC_C, KC_V, KC_B,
		KC_LCTL, KC_LEFT_GUI, KC_LALT, KC_APP, KC_SPC, KC_ENT, KC_PGUP,
		KC_SPC, KC_PGDN,
		KC_SPC, KC_INS,
        KC_F7, KC_F8, KC_F9, KC_F10, KC_F11, KC_F12, KC_DELETE,
        KC_EQL, KC_6, KC_7, KC_8, KC_9, KC_0, KC_BSPC,
        KC_RBRC, KC_Y, KC_U, KC_I, KC_O, KC_P, KC_BSLS,
        KC_QUOT, KC_H, KC_J, KC_K, KC_L, KC_SCLN, KC_ENT,
        KC_N, KC_M, KC_COMMA, KC_DOT, KC_SLASH, KC_RSFT,
        KC_HOME, KC_END,
        KC_UP, KC_SPC, KC_RALT, KC_RIGHT_GUI, KC_SPC, KC_RCTL,
        KC_LEFT, KC_DOWN, KC_RIGHT
    )
};
```

rules.mk:
```
BOOTMAGIC_ENABLE = yes      
MOUSEKEY_ENABLE = yes     
EXTRAKEY_ENABLE = yes     
CONSOLE_ENABLE = no        
COMMAND_ENABLE = no        
NKRO_ENABLE = yes        
BACKLIGHT_ENABLE = no       
RGBLIGHT_ENABLE = no        
AUDIO_ENABLE = no        
ENCODER_ENABLE = yes
SPLIT_KEYBOARD = yes
AUDIO_SUPPORTED = no
BACKLIGHT_SUPPORTED = no
RGBLIGHT_SUPPORTED = no
```

fauxpark, 2023-10-20:
Please read the documentation. Don't just enable features without understanding how they work. You've turned on Encoder but have not specified any pins for it.

Harrycheng1992, 2023-10-20:
> Please read the documentation. Don't just enable features without understanding how they work. You've turned on Encoder but have not specified any pins for it.

I understand, thank you very much for your reply

closed 2023-11-01 by zvecr