qmk/qmk_firmware #22335: [Bug] Mysterium got stuck and send keystroke repeatedely

ThunderrGodd, 2023-10-25:
### Describe the Bug

1. It often got stuck and ignores keystroke. it keeps 1~2 seconds.
2. If not ignoring keystroke, it sends the last keystroke repeatedely. the keystroke is oftenly multiple of 17 or 18.
3. It seems not periodic, but it happens suddenly.

### Keyboard Used

coseyfannitutti/mysterium

### Operating System

Windows 10 Home

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: C:/Users/Thundergodd/Documents/GitHub/qmk_firmware
Ψ Detected Windows 10 (10.0.19044).
Ψ QMK MSYS version: 1.7.2
Ψ Git branch: master
Ψ Repo version: 0.22.10
⚠ Git has unstashed/uncommitted changes.
⚠ The official repository does not seem to be configured as git remote "upstream".
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.1.0
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.0
Ψ Found dfu-programmer version 0.7.2
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-07-17 11:39:05 +0200 --  (da78eb37)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Additional Context

NKRO = yes
polling rate = 125hz
2 Langs on OS, but the problem happens in any language

zvecr, 2023-10-25:
https://github.com/qmk/qmk_firmware/issues/11389#issuecomment-1145083022
https://github.com/qmk/qmk_firmware/pull/17034
https://github.com/qmk/qmk_firmware/issues/15400#issuecomment-1399675457

ThunderrGodd, 2023-10-26:
after adding the code, there's no problem now. Thank you!

closed 2023-10-26 by ThunderrGodd