qmk/qmk_firmware #22348: [Bug] Cannot start nix-shell on macOS

synecdokey, 2023-10-27:
### Describe the Bug

Cloning the repository, then trying to start `nix-shell` results in the following errors:

```
error:
       … while calling the 'derivationStrict' builtin

         at /builtin/derivation.nix:9:12: (source not available)

       … while evaluating derivation 'qmk-firmware'
         whose name attribute is located at /Users/emiliazapata/projects/qmk_firmware/shell.nix:69:3

       … while evaluating attribute 'AVR_ASFLAGS' of derivation 'qmk-firmware'

         at /Users/emiliazapata/projects/qmk_firmware/shell.nix:82:3:

           81|   AVR_CFLAGS = lib.optional avr avr_incflags;
           82|   AVR_ASFLAGS = lib.optional avr avr_incflags;
             |   ^
           83|   shellHook = ''

       (stack trace truncated; use '--show-trace' to show the full trace)

       error: function 'anonymous lambda' called without required argument 'lib'

       at /nix/store/gm4rp1786ligd81v2qz420nkrwy0r50k-nixpkgs-src/pkgs/build-support/cc-wrapper/default.nix:8:1:

            7|
            8| { name ? ""
             | ^
            9| , lib
```

### Operating System

macOS 14.0

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Additional Context

nix (Nix) 2.17.0

sigprof, 2023-10-28:
The only way I can test the macOS support is by using the free GitHub runners, and apparently the current `shell.nix` works there (e.g., the latest run at the moment is https://github.com/sigprof/qmk-nix-support/actions/runs/6674858226/job/18142204580).  But the macOS version used by GitHub for `macos-latest` runners is apparently still 12.7; the Nix version is the default for `cachix/install-nix-action`, which is 2.17.0 at the moment.

However, if your Mac is actually based on Apple Silicon, the existing `shell.nix` probably won't work (in fact, I can reproduce a similar evaluation error by trying to run `nix-shell --system aarch64-darwin`).  The Nixpkgs snapshot that's used currently is probably too old to have proper `aarch64-darwin` support.

I also have some code that uses a more recent Nixpkgs snapshot (https://github.com/sigprof/nix-devenv-qmk); apparently the evaluation for `aarch64-darwin` also fails with that code, but with a much more clearer error message:
```
error: Package ‘avr-stage-final-gcc-wrapper-8.5.0’ in /nix/store/apfgz0vlvsmlifhqh6x0k2aih1jw3ikz-nixpkgs-src/pkgs/build-support/cc-wrapper/default.nix:628 is not available on the requested hostPlatform:
         hostPlatform.config = "aarch64-apple-darwin"
         package.meta.platforms = [
           "i686-cygwin"
           "x86_64-cygwin"
           "x86_64-darwin"
           "i686-darwin"
           "aarch64-darwin"
           "armv7a-darwin"
           "i686-freebsd13"
           "x86_64-freebsd13"
           "x86_64-solaris"
           "aarch64-linux"
           "armv5tel-linux"
           "armv6l-linux"
           "armv7a-linux"
           "armv7l-linux"
           "i686-linux"
           "loongarch64-linux"
           "m68k-linux"
           "microblaze-linux"
           "microblazeel-linux"
           "mips-linux"
           "mips64-linux"
           "mips64el-linux"
           "mipsel-linux"
           "powerpc64-linux"
           "powerpc64le-linux"
           "riscv32-linux"
           "riscv64-linux"
           "s390-linux"
           "s390x-linux"
           "x86_64-linux"
           "aarch64-netbsd"
           "armv6l-netbsd"
           "armv7a-netbsd"
           "armv7l-netbsd"
           "i686-netbsd"
           "m68k-netbsd"
           "mipsel-netbsd"
           "powerpc-netbsd"
           "riscv32-netbsd"
           "riscv64-netbsd"
           "x86_64-netbsd"
           "i686-openbsd"
           "x86_64-openbsd"
           "x86_64-redox"
         ]
         package.meta.badPlatforms = [
           "aarch64-darwin"
         ]
       , refusing to evaluate.
```
So it seems that the underlying reason for the error is that at least the AVR cross toolchain is broken on `aarch64-darwin`.  Adding `--arg avr false` may help if you don't actually need AVR support.  Another thing you may try is changing `pkgsCross.avr.buildPackages.gcc8` to `pkgsCross.avr.buildPackages.gcc` (this way you would get avr-gcc 12.3.0, which may produce larger binaries than 8.5.0, but it might be good enough).

And yet another option that you may try is to use the Rosetta emulation: `nix-shell --system x86_64-darwin`.

synecdokey, 2023-10-28:
I am indeed on apple silicon! `--arg avr false` is something I tried, but didn't work out. But using your devenv (with `--arg avr false`) worked like a charm, many thanks for getting me unblocked. ☺️

sigprof, 2024-02-24:
Nix shell updates in #23143 should fix this problem (although I don't have the hardware to actually test that, recently GitHub started to provide M1-based runners for `macos-14`, and the new `shell.nix` code seems to work there).

closed 2024-02-26 by sigprof