qmk/qmk_firmware #22349: [Bug] Repeat key won't work?

AaronKutch, 2023-10-27:
### Describe the Bug

The QMK configurator and local qmk_firmware have given me no problems so far besides needing to chase down the alternate names that some distros use for `arm-none-eabi-gcc`. I have a `beekeeb/piantor` board on which quantum keycodes like layers and adding on modifiers work. However, when I tried to add the repeat key QK_REP, the repeat key seems to do nothing. Both on the online QMK configurator and with the local build, and on a Linux and a Windows machine, it doesn't seem to do anything. I have attached my QMK layout that I want to get working, but editing the default to have a QK_REP anywhere also has the same effect.

[beekeeb_piantor_layout_split_3x6_3_rust_graphite_v10.json](https://github.com/qmk/qmk_firmware/files/13193833/beekeeb_piantor_layout_split_3x6_3_rust_graphite_v10.json)

### Keyboard Used

beekeeb/piantor

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

zvecr, 2023-10-28:
From <https://docs.qmk.fm/#/feature_repeat_key?id=how-do-i-enable-repeat-key>:

> In your rules.mk, add:

> > REPEAT_KEY_ENABLE = yes

Which is not available within the configurator.

AaronKutch, 2023-10-28:
where is this rules.mk? Also, why does this separate configuration exist in the first place? Can't compilation detect if QK_REP exists and then use the repeating functionality (and the configuration can be enabled by default, disabled by that flag in case someone needs it for whatever reason).

zvecr, 2023-10-28:
> Also, why does this separate configuration exist in the first place?/

It is one of the fundamental building blocks on QMK, from before the existence of `keymap.json`.

> and the configuration can be enabled by default

This is not the strategy that QMK follows for a few reasons. Primarily due to firmware size issues.

> Can't compilation detect if QK_REP exists and then use the repeating functionality

It *could* in the future, and a partial implementation has be mocked up a long while back. However some features require additional configuration which complicates a generic solution.

AaronKutch, 2023-10-28:
In case anyone else encounters this, I had to:
`python3 -m pip install --user qmk`
`qmk setup`
`sudo apt install gcc-arm-none-eabi dfu-util dfu-programmer` for my system
`qmk setup` until there are no errors
`qmk compile -kb beekeeb/piantor -km default` for testing
`qmk new-keymap -kb beekeeb/piantor` and enter some name without spaces
Open `.../qmk_firmware/keyboards/beekeeb/piantor/keymaps/[name]/keymap.c` and edit it with keycodes from the QMK Configurator, place QK_REP where I wanted, add more layers, etc.
Put a rules.mk file with just the line `REPEAT_KEY_ENABLE = yes` in the same directory.
`qmk compile -kb beekeeb/piantor -km [name]`
use the .utf2 file from `qmk_firmware/.build/...`

closed 2023-10-28 by AaronKutch