qmk/qmk_firmware #22394: [Bug] Fresh setup can't compile kinesis/kint41:default

cub-uanic, 2023-11-03:
### Describe the Bug

Did a fresh setup with all defaults and tried to compile.

```
% qmk compile -kb kinesis/kint41 -km default
Ψ Compiling keymap with gmake --jobs=1 kinesis/kint41:default

QMK Firmware 0.22.14
Making kinesis/kint41 with keymap default

Generating: .build/obj_kinesis_kint41_default/src/info_deps.d                                       [OK]
platforms/chibios/platform.mk:104: lib/chibios/os/hal/ports/MIMXRT1062/MIMXRT1062/darwin.mk: No such file or directory
gmake[1]: *** No rule to make target 'lib/chibios/os/hal/ports/MIMXRT1062/MIMXRT1062/darwin.mk'.  Stop.
gmake: *** [Makefile:392: kinesis/kint41:default] Error 1
```

### Keyboard Used

kinesis/kint41

### Operating System

macOS 12.7.1 (Intel)

### qmk doctor Output

```text
% qmk doctor
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.2
Ψ QMK home: /Users/cub/qmk_firmware
Ψ Detected macOS 12.7.1 (Intel).
Ψ Git branch: master
Ψ Repo version: 0.22.14
Ψ - Latest master: 2023-11-02 09:36:56 -0700 (837cae95ba) -- boardsource/unicorne - Fixed OLED layers being all set to zero (#22389)
Ψ - Latest upstream/master: 2023-11-02 09:36:56 -0700 (837cae95ba) -- boardsource/unicorne - Fixed OLED layers being all set to zero (#22389)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2023-11-02 09:36:56 -0700 (837cae95ba) -- boardsource/unicorne - Fixed OLED layers being all set to zero (#22389)
Ψ - Common ancestor with upstream/develop: None
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.5.0
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.2
Ψ Found dfu-programmer version 1.1.0
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-07-17 11:39:05 +0200 --  (da78eb37)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

tzarc, 2023-11-03:
I think this is somehow environmental -- seeing `darwin.mk` in the output is plain wrong -- have you got any environment variable on your machine called `PLATFORM_NAME`?
```sh
echo $PLATFORM_NAME
```
I'm guessing _somehow_ this is set to `darwin` and QMK's build system is taking the value from the environment.

tzarc, 2023-11-03:
You can also try running:
```sh
unset PLATFORM_NAME
```
...then attempting to compile again.

cub-uanic, 2023-11-03:
Yes, you were right, there was `$PLATFORM_NAME`, and unsetting it helped.
Thanks a lot!

closed 2023-11-03 by cub-uanic