qmk/qmk_firmware #22450: [Bug] Audio doesnt work on Nullbits Snap

freemo, 2023-11-11:
### Describe the Bug

I have been trying for a day to get the speaker working on my nullbits snap. It compiles fine but nothing I do can get it to work.

Here is my source code and setup, it is forked from the official snap repo for rp2040: https://git.qoto.org/freemo/qmk-firmware/-/tree/freemo/keyboards/nullbitsco/snap/rp2040/keymaps/freemo?ref_type=heads

I tried to merge in master and it didnt fix anything, though i tried it first without merging in. Also the current branch linked has AUDIO_ENABLE = no, Obviously when i try it with audio i enable this.

While I was able to find the documentation for audio on the QMK official firmware I couldnt find anything from nullbits. I did however speak to a nullbits Dev and they assure me the hardware supports sound. Sound doesnt play but otherwise everything works. 

I tried setting keys to toggle audio, and clicky and music though it doesnt seem to do anything either. I tried hooking up a oscilloscope to the audio pin, pin GP21 and then turn on clicky with the CK_TOGG key. Then i tried typing to look for any activity on the audio pin... I see nothing. There is also nothing that shows up when it powers up that might suggest a startup sound. As best I can tell the firmware just silently refuses to output the audio on the specified pin.

### Keyboard Used

Nullbits Snap

### Link to product page (if applicable)

https://nullbits.co/snap/

### Operating System

NixOS - Linux

### qmk doctor Output

```text
❯ qmk doctor                   
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.2
Ψ QMK home: /home/freemo/source/qmk_firmware
Ψ Detected Linux (NixOS 23.11 (Tapir)).
⚠ Missing or outdated udev rules for 'atmel-dfu' boards. Run 'sudo cp /home/freemo/source/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'kiibohd' boards. Run 'sudo cp /home/freemo/source/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'stm32-dfu' boards. Run 'sudo cp /home/freemo/source/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'apm32-dfu' boards. Run 'sudo cp /home/freemo/source/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'gd32v-dfu' boards. Run 'sudo cp /home/freemo/source/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'wb32-dfu' boards. Run 'sudo cp /home/freemo/source/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'bootloadhid' boards. Run 'sudo cp /home/freemo/source/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'usbasploader' boards. Run 'sudo cp /home/freemo/source/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'usbtinyisp' boards. Run 'sudo cp /home/freemo/source/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'md-boot' boards. Run 'sudo cp /home/freemo/source/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Detected ModemManager without the necessary udev rules. Please either disable it or set the appropriate udev rules if you are using a Pro Micro.
⚠ Missing or outdated udev rules for 'caterina' boards. Run 'sudo cp /home/freemo/source/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'hid-bootloader' boards. Run 'sudo cp /home/freemo/source/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
Ψ Git branch: freemo
Ψ Repo version: 0.22.14
Ψ - Latest freemo: 2023-11-11 13:10:11 -0500 (8c51a438c3) -- Added some guards to make it compile
Ψ - Latest upstream/master: 2023-11-09 04:03:07 +0200 (daabe2d8c5) -- Update layer state of Lily58 from 'Default' to 'Base' (#22397)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2023-11-09 04:03:07 +0200 (daabe2d8c5) -- Update layer state of Lily58 from 'Default' to 'Base' (#22397)
Ψ - Common ancestor with upstream/develop: None
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 12.2.1
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.2
Ψ Found dfu-programmer version 0.7.2
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-07-17 11:39:05 +0200 --  (da78eb37)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

zvecr, 2023-11-11:
> #define AUDIO_PWM_DRIVER PWMD1
> #define AUDIO_PWM_CHANNEL 1

This is most certainly wrong for GP21.

Referencing <https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf#page=525&zoom=100,153,620>,

```c
#define AUDIO_PWM_DRIVER PWMD2
#define AUDIO_PWM_CHANNEL RP2040_PWM_CHANNEL_B
```

freemo, 2023-11-11:
> ```c
> #define AUDIO_PWM_DRIVER PWMD2
> #define AUDIO_PWM_CHANNEL RP2040_PWM_CHANNEL_B
> ```

Thanks, I will try this now. Should have a response to you within the hour (originally i left those settings off, I am not sure what it defaulted to).

zvecr, 2023-11-11:
> `#define AUDIO_PWM_PAL_MODE 42`

This should also be removed.

freemo, 2023-11-11:
> > `#define AUDIO_PWM_PAL_MODE 42`
> 
> This should also be removed.

Thanks, that got me most of the way there. Once I removed my init code from main and recompiled sound is now working. Thank you!

closed 2023-11-11 by freemo